FreeCAD/FreeCAD #22586: PartDesign: Hole on a Padded Involute Gear Freezes the Program

back-ref, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi! I was working on a design learning FreeCAD and noticed when using Hole on a padded InvoluteGear it freezes my program for 5+ minutes.

**Specs:**
-   Ryzen 7 5700G
-   RTX 3060 12GB
-   32GB 3200MHz RAM
<br/>

**Replication:**

1. Insert InvoluteGear inside a Body with default parameters(any parameter causes freezes for me).
2. Pad the gear and then attempt to use Hole on the top or bottom face.
<br/>
My CPU hits 100% usage, temps rise and the program is no longer useable.
I can wait 5+ minutes for it to unfreeze or close it and lose progress.

[image]

Here is the result of letting it unfreeze itself:

[image]

Luckily pressing redo does not take 5+ minutes if I let it unfreeze.

### Full version info

```shell
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

The freezes occur in safe mode as-well.
```

### Subproject(s) affected?

PartDesign

### Anything else?

I hope I'm not making a noob error :c.
Any advice is appreciated!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

back-ref, 2025-07-19:
Just came up with a solution for my issue!
Used Grok to make button that works similar to hold but applies a circle sketch prior to the hole which stops the program from freezing!

https://gist.github.com/back-ref/88c3f0862057cd3172d2dcb1d81a336a

drwho495, 2025-07-20:
stalls for a while on linux
[image]

the stalling is caused by opencascade
which also explains why multiple cores jump in usage, not just one

drwho495, 2025-07-20:
ok so this is kind of a duplicate of https://github.com/FreeCAD/FreeCAD/issues/22595