FreeCAD/FreeCAD #22588: Assembly: revolute joint give an error when used two times in one part

muezabdalla, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

when trying to make this small assembly

[image]

I managed to assembly the first two parts by just two revolute joint for each one 
but for the 3rd one the first joint is done but when selecting the outer edge in the base and the inner edge of the slot hole and press revolute joint like this:

[image]

I get this error:
```
2.8e-08 <Assembly> AssemblyObject.cpp(160): Solve failed: vector::_M_range_check: __n (which is 84) >= this->size() (which is 84)
```

[revolute_joint.zip](https://github.com/user-attachments/files/21327417/revolute_joint.zip)

### Full version info

```shell
OS: Void Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.42594 (Git)
Build date: 2025/07/17 18:40:08
Build type: Unknown
Branch: main
Hash: 3f6a3f10ffd617369fa9a12d23496f663c44a855
Python 3.13.5, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.1, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.38
  * AnimationFreeCAD 1.0.0
  * FreeCAD_Assembly4.1 0.50.18
  * ExplodedAssembly
  * CfdOF 1.31.4
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

muezabdalla, 2025-07-19:
is the Assembly solver strict about the overconstrainted like the stuff like the Assembly3 solver or it is like A2+ solver.
because in A2+ this would be fine.

secondly: why in the first two parts it did not give any errors?

and I check the dimensions and did not find any thing wrong with it
can you tell what exact dimension is wrong?

drwho495, 2025-07-19:
> is the Assembly solver strict about the overconstrainted like the stuff like the Assembly3 solver or it is like A2+ solver. because in A2+ this would be fine.
>
> secondly: why in the first two parts [...]

its not as strict but weird stuff can happen. you might need to delete and recreate your joints on the broken part in different orders