FreeCAD/FreeCAD #22594: Unable to open groups (folders)

MartyDavis, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Newly created or existing groups cannot be opened. 

Start a new file, create a body and then create a group. Place the body into the group. Try to open the group; there is no means to do so. Existing groups with objects inside cannot be opened.

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/108/2
Installed mods: 
  * DynamicData 2.74.0
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-07-20:
> Start a new file, create a body and then create a group. Place the body into the group. Try to open the group; there is no means to do so. Existing groups with objects inside cannot be opened.

Double click does not open the group. Where it used to be a dropdown icon, it's not there anymore, but the functionality remains. Click in that sweet spot will open the group, but the icon is missing.

[image]

Katt-Kushol, 2025-07-20:
All the dropdown icon are gone. 

[image]

Version:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.64
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.24
  * Silk 0.2.0

Roy-043, 2025-07-20:
https://github.com/FreeCAD/FreeCAD/issues/22476

closed 2025-07-21 by maxwxyz