FreeCAD/FreeCAD #22599: Safe height and Clearance height in operations - Safe height ignored.

Connor9220, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Safe height isn't being used in most of the operations.

We have lots of wasted Z movements with inner linking between features in the same operation. (Having to go up to clearance height, move to next feature and then back down to resume next feature).

Drilling Operation has Retract Height that is linked to KeepToolDown which is confusing.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * AddonManager 2025.6.29
  * lattice2.backup1751140073.422284 1.0.0 (Disabled)
  * lattice2 1.0.0
  * ThreadProfile.backup1751140075.2052586 1.92.0 (Disabled)
  * ThreadProfile 1.96.0
  * A2plus 0.4.68
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * OpenTheme 2025.5.20
  * fasteners.backup1725502068.289526 0.5.24 (Disabled)
  * A2plus.backup1725502065.0763218 0.4.65 (Disabled)
  * fasteners 0.5.38
  * dodo 1.0.1
  * QuickMeasure 2022.10.28
  * ThreadProfile.backup1725502069.7449253 1.90.0 (Disabled)
```

### Subproject(s) affected?

CAM

### Anything else?

To kick things off on how to deal with this, here is some pseudo code we went over in a work session. 

```
# Purposed function to be used in a linking generator to provide height information for a given operations inner linking moves between features. 
# Introduces two (three if we choose to add threshold linkMode) new attributes to most operations
# linkMode: Which mode to use, alwaysClearanceHeight, alwaysSafeHeight, threshold, auto
# retractHeight: Height Offset to use for a inner linking move between two features in a given operation.
# thresholdDistance: Max distance between two features in a given operation where we "assume" it's safe to use the retractHeight vs clearanceHeight

# For drilling operations, we need to have a retractHeight, a safeHeight and clearanceHeight
# innerOpLinking is used to select the height to which the tool moves before advancing to the next hole
# The clearanceHeight is the height to which the tool moves before advancing to the next operation

# This means all operations will have a retract height to be consistent. 

# ClearanceHeight - Clears all clamps and fixtures
# -== This zone is the space for clamping ==-
# SafeHeight - is height above the top of the stock that is safe to move the tool at rapid speeds for the current operation not takinginto account innerOpLinking and collisions with the stock (fixturing not taken into account)
# RetractHeight is individual hole or pocket startDepth + setupsheet.RetractHeightOffset
# StartDepth (defaults to opStartDepth)
# StepDown
# FinishDepth
# FinalDepth (defaults to opFinalDepth)

innerOpLinking(startx,starty,endx,endy,linkMode)
    if linkMode == alwaysClearanceHeight # User override to always move tool at clearanceHeight for inner operation geometries linking
        return clearanceHeight
    elseif linkMode == alwaysSafeHeight # User override to always move tool at safeHeight for inner operation geometries linking
        return safeHeight
    elseif linkMode == alwaysRetractHeight # User override to always move tool at retractHeight for inner operation geometries linking
        return retractHeight
    elseif linkMode == threshold # Threshold mode - Stop gap method before implmenting collision and fixtureCollision
        if distance < thresholdDistance
            return retractHeight
        else
            return safeHeight
    elseif  linkMode == auto    # Auto mode
        if not collision(startx,starty,endx,endy,retractHeight)
            return retractHeight
        elseif not fixtureCollision(startx,starty,endx,endy,safeHeight) # (next level CAM)
            return safeHeight # return safeHeight if we do not add fixtureCollision any time soon

    return clearanceHeight

collision(startx,starty,endx,endy,height)
    # Check if the path from (startx, starty) to (endx, endy) at the given height collides with any obstacles
    # We have a to decide if this is going to be a stock collision or a model collision (Start with model collision)
    # Stock collision is when the tool path collides with the stock material that may have not been removed yet (next level CAM)
    # Model collision is when the tool path collides with the model geometry (stock could still be there)
    return false  # Assume no collision for this example

fixtureCollision(startx,starty,endx,endy,height)
    # Check if the path from (startx, starty) to (endx, endy) at the given height collides with any fixtures
    # This is a placeholder function; actual implementation would depend on the specific environment and fixture detection logic
    return false  # Assume no fixture collision for this example```

# Credits: sliptonic, Connor, jffmichi, MTronic, Jannek``

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

J-Dunn, 2025-07-26:
> Drilling Operation has Retract Height that is linked to KeepToolDown which is confusing.

There is a necessity for an additional height in drilling operations. Drilling ops themselves G81,82,83 etc. define a "retract plane".  This is independent of what FreeCAD does between cuts or into and out of ops. Though under certain options the behaviour of drilling ops is affected by the machine position when the G8* command is sent. 
See https://github.com/FreeCAD/FreeCAD/issues/22622

This is separate from the main issue of the height of rapids between cuts in the same op.  and the lack of G0 from clearance down to "safe" height before cutting. 

Somewhere it seems changes were made which made "safe height" almost without use and seriously degraded performance.   This may have even happened before v1.0.0

sliptonic, 2025-07-29:
> There is a necessity for an additional height in drilling operations. Drilling ops themselves G81,82,83 etc. define a "retract plane". This is independent of what FreeCAD does between cuts or into [...]

ADR-003 (DRAFT)  sets out clarification about heights.  It says that operations _may_ define additional height planes to be used within the operation.  That seems to be consistent with a retract plane.

I'll propose that we start discussing two pieces of code:

# Linking Generator

## Assumptions
- The user may or may not have a 3D model and stock.  They may be working from 2D base object like an imported DXF.
- The calling function may want to test for collision against model, stock, both, or none.
- The calling operation (or job) may use the resulting moves directly or may override them.  
- local_clearance must be lower than global clearance.
- Detecting IF there's a collision against whatever is passed is sufficient.  We don't have to report back on WHICH solid had the collision.
- If the start_position.z is the same as the target_position.z, retraction is not needed unless collision is detected.

```
import Part
from FreeCAD import Vector
from typing import Optional

def get_linking_moves(
    model: Optional[object] = None,
    stock: Optional[object] = None,
    local_clearance: Optional[float] = None,
    global_clearance: float = 5.0,
    start_position: Optional[Vector] = None,
    target_position: Optional[Vector] = None,
) -> list:
    if not start_position or not target_position:
        raise ValueError("start_position and target_position must be provided")

    if start_position == target_position:
        return []

    if local_clearance is None:
        local_clearance = global_clearance

    if local_clearance > global_clearance:
        raise ValueError("Operation safe plane must not exceed job safe plane")

    shapes = [s.Shape for s in (model, stock) if s]
    collision_model = Part.makeFuse(shapes) if len(shapes) > 1 else shapes[0] if shapes else None

heights = []
seen = set()

for h in (
    start_position.z if start_position.z == target_position.z else None,
    local_clearance,
    global_clearance,
):
    if h is not None and h not in seen:
        heights.append(h)
        seen.add(h)

    wire = None
    for height in heights:
        trial_wire = make_linking_wire(start_position, target_position, height)
        if is_wire_collision_free(trial_wire, collision_model):
            wire = trial_wire
            break

    if not wire:
        raise RuntimeError("No collision-free path found between start and target positions")

    cmds = Path.cmdsForWire(wire)
    return cmds

def make_linking_wire(start: Vector, target: Vector, z: float) -> Part.Wire:
    p1 = Vector(start.x, start.y, z)
    p2 = Vector(target.x, target.y, z)
    e1 = Part.makeLine(start, p1)
    e2 = Part.makeLine(p1, p2)
    e3 = Part.makeLine(p2, target)
    return Part.Wire([e1, e2, e3])

def is_wire_collision_free(wire: Part.Wire, solid: Optional[Part.Shape], tolerance: float = 0.001) -> bool:
    if not solid:
        return True
    distance = wire.distToShape(solid)[0]
    return distance >= tolerance

```

# Unit test suite

- It's an error if the local height is above the global height
- The generator should return an error if no path is found
- It's an error if the start point and end point are the same
- The result should be a set of commands.

sliptonic, 2025-07-29:
Is this assumption valid?
- If the start_position.z is the same as the target_position.z, retraction is not needed unless collision is detected.

If we're finishing a pocket of one target and moving to another, we have done no vertical move yet.  But the target.z should be higher because we'll start above the next target and feed into it.

If we're drilling, the previous command in the cycle will have retracted to the retraction plane so should be safe to transit to the next target laterally only.

Under what conditions would the assumption NOT be valid?

sliptonic, 2025-07-29:
We can make the generator simpler by always retracting to the local plane.  
But as @J-Dunn has pointed out, the drilling operation is weird.  Within the context of the canned cycle, there is no explicit move from hole to hole.  It's handled by the canned cycle.

So I'm kind of going back and forth trying to think about how the drilling operation would interact with the linking generator.

I assume that the operation will call the generator for each target.  If a REAL linking move is necessary, for example to avoid a collision, it will end the cycle, use the linking move, and start the next cycle.

But if a REAL linking move isn't needed, it will throw away any linking move and just adds the next hole in the same cycle.

Whatever the generator returns, it has to be semantically rich enough for the drill op to make the right decision.

Maybe you have a better suggestion for a return type that the drilling operation could use to decide its next action.

jffmichi, 2025-07-29:
Let me answer in two parts:

In my mind, the whole "drilling operation is special" is not true. Only because the GCode standard defines two different retract modes we don't have to use those modes. Ideally the user will not care about the GCode, it's just an intermediate format to tell the machine what to do. If we forget about the "GCode standard" for a moment, the drilling operation is just like any other operation: move to feature A, do something, linking move to freature B, do something, ... (where "something" is defined by the operation.

Now for the actual problem: I think returning the already prepared linking path makes it easy for all operations to just append that and continue with their business, that's good. Maybe return the prepared linking path and the height? That should be enough information for the drilling op to decide if it want's to use the prepared linking path or stay in the canned cycle (if the returned height is <= retract height)

J-Dunn, 2025-07-29:
> If we're drilling, the previous command in the cycle will have retracted to the retraction plane so should be safe to transit to the next target laterally only.
> 
> Under what conditions would the assumption NOT be valid?

consider peck drilling:

G90
G0 Z0 X0 Y0
G83 Z-10 R-2 P2 G98  ; old_z retraction mode: ie z0
X5 Y5 ; repeat canned cycle at new location
G80

G90
G0 Z0 X0 Y0
G83 Z-10 R-2 P2 G99  ; R plane retraction mode: ie z-2
X5 Y5 
G80

If z=0 is top of stock , the former will remain inside the hole: R-2, only retracting to Z0 at end.
Avoiding multiple re-entry can be a desirable feature. 
https://www.catking.net/cnc-img/peck68.mp4  # be patient, 10s wordpress sluggishness.

The second, while legit Gcode will do final retraction to z-2 and snap the drill. 
If stock is at -2, this would rapid to -2, do the peck, retracting to -2 each peck ending at -2.

If, and only if, Z is below R, at rapid will be done to **_raise_** to R before initial XY rapid. 
Cutting will start directly on starting G8x, there is no rapid **_down_** to R . 

Retract plane has a defined meaning for drilling ops ( and meaning of R param can vary according to  G98/99). 
As such we should probably avoid using "Retract plane" to mean something else in FreeCAD. 
NIST uses "clear Z" for the final move and this is adopted by several Gcode interpreter descriptions of drilling ops.
The "clear Z" may or may not be equal to retract height, dependent on G98/99.

In earlier discussion we saw some other software using Rapid Height/Plane or Feed Height/Plane for the locally safe height at which downward G0 end before cutting and at whcih rapids between cuts of the same op. occur.

sliptonic, 2025-07-29:
> In my mind, the whole "drilling operation is special" is not true. Only because the "GCode standard" defines two different retract modes we don't have to use those modes. Ideally the user will not care about the GCode, it's just an intermediate format to tell the machine what to do. If we forget about the "GCode standard" for a moment, the drilling operation is just like any other operation: move to feature A, do something, linking move to freature B, do something, ... (where "something" is defined by the operation. Edit: and if we do the linking generator nicely it will still contain all the functionality the "GCode standard" provides.

That's an excellent point.

> 
> Now for the actual problem: I think returning the already prepared linking path makes it easy for all operations to just append that and continue with their business, that's good. Maybe return the prepared linking path and the height? That should be enough information for the drilling op to decide if it want's to use the prepared linking path or stay in the canned cycle (if the returned height <= retract height)

It's trivial to extract the height from the linking path.  So I prefer to return that.

J-Dunn, 2025-07-29:
I think the main reason for canned cycles is to prevent the need to repeat all the parameters if one or more are unchanged. 
This reduces byte count and may avoid transcription errors. If I have a set on holes aligned along Y I can define all I want it do in the G8x command then simply command a new hole with a single X coord. 

As jffmichi suggests , linking of drilling can be implemented the same as other ops by using G98 and explicitly doing an initial G0 down to where you want the cutting to start and G0 out to link height after each hole. 

There is still a need to and additional editbox for the R parameter in peck of chip break cycles.

J-Dunn, 2025-07-29:
> This is an advantage in the gcode but is meaningless in the intermediate path representation. Let's not confuse the two.

If there is the intent to use canned cycle features, it may be more robust to go straight there, rather than try to sniff them out and condense them later. 

On the contrary, it would require more thorough sims, these are currently severely challenged when it comes to drilling ops.

sliptonic, 2025-07-29:
I don't know about you guys but I keep getting twisted up because we're using words in an imprecise way.
You'll notice in my draft of the generator I don't use 'safe height' or 'clearance height' at all.  That's on purpose because those words have become so overloaded that they are almost meaningless now.

We're having the same problem with 'canned cycles'.  Canned cycles are a g-code thing.  At this point in the execution process (operations and generators) g-code and canned cycles are no where to be seen.  Let's banish that term from this discussion unless you're being **extremely** precise in what you mean.

Likewise, simulation discussion belongs in another topic not the one about the linking move generator.

The only discussion we need to have about drilling is whether this generator is giving the calling operation enough information to take the next step so it makes sense to talk about it a little bit.

J-Dunn, 2025-07-30:
> Likewise, simulation discussion belongs in another topic not the one about the linking move generator.

I did not intend to start discussing changes to sims, just note that changes here could impact the sims.  

> Let's banish that term from this discussion unless you're being extremely precise in what you mean.

What is the imprecision that is causing problems?  I thought my comments were accurate and precise.

> So I'm kind of going back and forth trying to think about how the drilling operation would interact with the linking generator.

I took that to be a call for comment on how that interaction could work and tried to address that. I also discussed to what extent that could be avoided. 

What did I get wrong?

sliptonic, 2025-07-30:
Canned cycles don't exist in the internal path representation.  
If you refer to the path commands as canned cycles, it's imprecise.  If you really are talking about gcode canned cycles, then unless you're extremely clear, it probably doesn't belong here at all.  Gcode generation is done at post processing and outside the scope of this discussion.

Not a criticism of just you. I'm guilty of it too.

J-Dunn, 2025-07-30:
OK.  Thanks. 

NIST calls all these G8x drilling cycles "canned cycles" , even if there is only one hole in the can. 
I guess the idea is that the "sticky" parameters are set up in the first G8x command and the cycle is defined even if you only run it once, as FreeCAD currently does. We should probably only use "canned cycles" to refer to Gcode since FC does not can anything. For FreeCAD they are drilling operations. 

>  Gcode generation is done at post processing and outside the scope of this discussion.

while we do not need to refer to G0,G1 etc. this discussion is about individual rapids and feeds and avoiding air cutting. While many 2.5D ops are composed of simple G1,G2,G3 cuts , if complex commands like the G8x series are used as well it seems necessary to have an understanding of where they leave the tool, to determine the minimum efficient moves between cuts.

I'm aware of two levels code production : 
- that produced by Post | Inspect Gcode  which always has F0 for drilling and helix which use FeedRate.py 
- and the final version after application of the machine specific post-proc scripts. 

I thought post processing referred to that final step. Are you including the first step in what you call post processing?
To my mind the first was processing and the second post-proc. but that is simply inferred and could be different from what you mean by these terms. Could you clarify how you use these terms?

> Canned cycles don't exist in the internal path representation.

By "internal path representation" are you referring to the data structure of variables associated with the UI and the properties table, like Start Depth, Final Depth, Step Down, Step Over, Dwell Time, Peck Depth, etc.? 
Where does the 2D clearance path fit in to this? 

Profile, pocket, milling all have a 2D plan of action executed at a progressive series of Z heights.
Drilling, Helix ops also have a 2D plan of points, however in these cases each point needs to process the Z plan fully before moving in XY. 

So drilling/helix ops are a series of pockets with a single point clearance pattern.  However, all the detail is in the much less trivial Z plan. They do not need dress-ups to finish the task. 

If you want to avoid all talk of G8x and keep if simply functional, this leads to decomposing peck drilling into a Z plan of rapids and cuts (G0,G1). This level of abstraction should be possible but would massively increase the volume of Gcode output and make manual verification a LOT harder.  It seems foolish not to leverage machines which already fully provide drilling cycles. ( That's not my case BTW, I'm grbled. )

I see the motivation for separating function from Gcode and I like the idea of abstraction to a "grand unified theory" of 2.5D ops, from milling to drilling, but it has its downsides .

jffmichi, 2025-07-30:
A thousand or more comments into this discussion and we're still discussing terminology... 🤦

This thread is about movement between features in an operation. Everything that contacts material needs to disappear from this thread. Let me repeat myself, every operation does the following:
```
Move to feature A
do "something"
Linking move to feature B <---- this and only this is what the linking generator is about
do "something"
....
Move up to job safe plane
```

"something" is defined by the operation. It doesn't matter if it's cutting material sideways, drilling downwards (with or without peck), fiddling with z plans, rolling on the floor or going through a black hole and loop around the universe.

The only two things that matter are:

- where did the operation finish feature A
- where does it want to start feature B

Those are defined by the operation. What they are for which operation doesn't matter. The operation needs to tell those coordinates to the linking generator and the linking generator will provide a path from A to B.

J-Dunn, 2025-07-30:
> The only two things that matter are:
> 
>     * where did the operation finish feature A
> 
>     * where does it want to start feature B

Good approach. 

This probably leads to removing the user entered "safe" height, retaining only the globally safe job clearance (which manually accounts for fixation without needing to model them in). 

Each cut feeds back its end point and then robust collision detection with the 3D model gets it to where it needs to go next.

jffmichi, 2025-07-30:
We DEFINE what the internal path representation drilling commands are. And we seem to have defined them to behave similar to the similarly named GCode canned cycles, i.e. imply that lateral move. See comment here
https://github.com/FreeCAD/FreeCAD/blob/bc207c0a077e6db166cf36033ccf72042a85216f/src/Mod/CAM/Path/Op/Drilling.py#L276-L277
But yes I agree we should make that definition explicit either way and add it to the wiki.

We can never assume perfect collision detection, this is why we need a way for the user to override.

jffmichi, 2025-07-30:
We can never assume perfect collision detection. But even if we could, the user would still need to model all the clamps or stuff. They might not want to do that because they're only drilling 2 holes and don't care that the linking move is not optimal.

We need a way for the user to override the collision detection. (!!!!) 

(edit: I'll add that to the "only things that matter" section above to make it clearer)

J-Dunn, 2025-07-30:
>  internal path representation

Could someone explain what this refers to ?

jffmichi, 2025-07-30:
> > internal path representation
> 
> Could someone explain what this refers to ?

https://wiki.freecad.org/CAM_scripting#The_FreeCAD_Internal_GCode_Format

That's the output of the operations, i.e. what you see when you do "Inspect Path", i.e. the input to the postprocessors.

jffmichi, 2025-07-30:
I would argue to put that user override into the generator. This forces calling code to be consistent and the generator can already output the correctly prepared linking path. But you're actually working on it so your choice ofc.

Vote for option one to simplify using the generator. Otherwise we duplicate the union stuff at every callsite. The more functionality that can be offloaded from the callsite the better.

J-Dunn, 2025-07-30:
> Option one, the generator accepts a list of solids to test collision against. It does whatever it likes. Probably union together.

Does that potentially mean both stock and model collision? 
Is the token tool "cylinder" passed?

jffmichi, 2025-07-30:
> The generator has to generate correct linking moves for any operation.

An I'm sure we will see countless edgecases where we fail to do that. In order to keep the operation usable in those cases we should give the user a way to tell us: stop trying to be clever and just move at height X. We we're mostly agreeing on this when we wrote the pseudocode in the original post.

Not sure if I'm reading your [comment](https://github.com/FreeCAD/FreeCAD/issues/22599#issuecomment-3137136480) correctly: are you suggesting not to provide a user override at all or are you suggesting to handle the user override in the calling code by not passing in a solid to test against?

I can already see myself generating dummy geometry to get the linking generator to do what I want... 😝

sliptonic, 2026-10-15:
The main problem in this issue is solved, so I will close it. Here is a summary of what merged after the discussion above.

Linking generator
Operations now have one shared linking generator (Path/Base/Generator/linking.py). It makes the move from the end of one feature to the start of the next feature, and it can check for collisions with the model.
- #28180: collision detection uses the tool shape.
- #30399: the generator tests a list of candidate heights and uses the lowest safe height.
- #26195: fix for pure vertical linking moves.

User control
Each operation that uses the generator has the property CollisionAvoidanceStrategy in the Linking group. It has five options:
- Clearance Height: no collision detection. Linking moves go to the clearance height.
- Retract Height: no collision detection. Linking moves go to the safe height.
- Line of Sight: checks the centerline of the move.
- Tool Diameter: checks a clearance equal to the tool diameter.
- Tool Shape: checks the cross-section of the tool shape.

The two height options are the user override that we discussed above. The user can stop the collision detection and select a fixed height.

Operations that use the generator
- Profile and Pocket (#30402)
- Drilling, Helix and Tapping (#31602, #32281, #32780)
- Engrave and Deburr (#29959, #32114)
- Mill Facing (#31744)
- Flute

Drilling retract height
The R value of the drilling cycle is now the property PeckRetract. It does not have a connection to the linking between holes (#32281). The linking between holes is safe (#32780).

Heights terminology
The ADR for heights and depths merged in #25365.

Dressups
The dressups now keep the linking moves of the base operation, or they link the same way as the operation (#30402 and the PRs after it).

Not included
- Thread Milling: it shows the CollisionAvoidanceStrategy property but does not use it. Linking between holes always goes to the clearance height. I will open a separate issue for this.
- Adaptive, Slot, V-carve, 3D Surface and Waterline: these do not use the generator yet. Each one has its own path structure. If one of them makes linking moves that are not efficient, please open a new issue for that operation.
- Linking between operations: this is a different problem. It is part of the Job-level work for multi-axis machining.

Thank you to everyone who helped with the design here, and to tarman3 for most of the implementation.

closed 2026-10-15 by sliptonic