FreeCAD/FreeCAD #22609: Multi-Line preamble CAM Job --> Argumnents

PatrikRindlisbacher, 2025-07-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

--preamble="G17\nG90\nG64 P0.1"

These arguments are not resolved as multiple lines during refactoring post processor

### Full version info

```shell
FreeCAD Version 1.0.1 on Windows 10 /Windows 11
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

LarryWoestman, 2025-07-26:
Agree that this is a bug.  The code was fixed in the more recent "refactored" code base but the fix has not been backported to the 1.0.1 release.

In the 1.1 (ongoing development code) the corresponding line is:
```            values["PREAMBLE"] = args.preamble.replace("\\n", "\n")
```

LarryWoestman, 2025-07-26:
This fixes the issue at argument parsing time rather than later in the output process.

LarryWoestman, 2025-07-26:
The corresponding PR with this fix was #20833.

luzpaz, 2025-07-28:
Closing as fixed per @LarryWoestman

closed 2025-07-28 by luzpaz