FreeCAD/FreeCAD #22611: mirror point in draft result in the wrong position

ChristianKaspar, 2025-07-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Flipping a point does not result on the opposite side of the mirror axis. The point is generated on the end of the m

[image]

irror axis. In contrast, mirroring a line works correctly.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.18
  * BillOfMaterials 1.0.3.2
  * CurvedShapes 1.0.13
  * Curves 0.6.63
  * dodo 1.0.1
  * EasyProfileFrame 0.0.1
  * fasteners 0.5.38
  * FrameForge 0.1.3
  * freecad.gears 1.3.0
  * ImportNURBS 1.1.0
  * Manipulator 1.6.0
  * sheetmetal 0.7.23
  * ShortCuts
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-07-20:
Confirmed. The problem is related to [Part_Mirror](https://wiki.freecad.org/Part_Mirror) as that is the tool that is used here. It also occurs when mirroring a [Part_Point](https://wiki.freecad.org/Part_Point).

[part-mirror-issue.FCStd.zip](https://github.com/user-attachments/files/21335591/part-mirror-issue.FCStd.zip)
Remove the .zip extension.

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

luzpaz, 2025-12-06:
@ChristianKaspar can you reproduce on 1.2.0-dev (Weekly) using Roy's example ?  https://github.com/FreeCAD/FreeCAD/releases

webmite, 2026-01-03:
There is another test file you can use to verify functionality from this link [https://forum.freecad.org/viewtopic.php?p=865546#p865546](url)

It uses the Base and Normal reference so it allows you to test away from the Global Axis.

webmite, 2026-01-09:
Also tested above script successfully on all cases supplied in @Roy-043 test file 

[part-mirror-issue_imm1a.FCStd..zip](https://github.com/user-attachments/files/24525728/part-mirror-issue_imm1a.FCStd.zip)

The shape placement adjustment mirrors the observed behavior of the vertex in a part primitive circle, so
it should be accepted in most models.