FreeCAD/FreeCAD #22615: Part: Part_EditAttachment does not take nesting into account

Roy-043, 2025-07-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The Part_EditAttachment command does not take the nesting of the referenced object into account.

1. Open the attached [file](https://github.com/user-attachments/files/21335934/datums-attach-test.FCStd.zip) (remove .zip).
2. Select the Sketch.
3. Invoke Part > Attachment...
4. Pick DatumPlane001 in the 3D view.
5. Select the 'Plane face' Attachment mode.
6. Press the OK button.
7. Notice that the sketch has been realigned, but is not co-planar with the plane.
1. Reopen the file.
2. Repeat the test but now select 'Attachment editor' from the Tree view context menu of the sketch.
3. Notice that the 'Plane face' Attachment mode has already been selected.
4. Notice also that the sketch is aligned properly.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-21:
@PaddleStroke FYI

webmite, 2025-09-21:
I have modified the TaskAttachmentEditor file. 
It appears to resolve the above issue but requires further testing.

Thanks.

[TaskAttachmentEditor.py](https://github.com/user-attachments/files/22451646/TaskAttachmentEditor.py)

Roy-043, 2025-09-22:
Thanks for working on this. Your solution works for the given example. But not if you make the Line visible and then select that in the 3D View instead of the DatumPlane001 object.

I think you need to change the 1st part of `addSelection` to:
```python
    def addSelection(self, docname, objname, subname, pnt):
        # -- added for debugging. imm
        debug_pr(f'observer  doc:{docname}   obj:{objname}    sub:{subname}   pnt:{pnt}')
        sel = Gui.Selection.getSelectionEx("", 0)
        s0 = sel[0]
        objname = s0.Object.Name
        subname = s0.SubElementNames[0]
        # -- added for debugging. imm
```
It's a bit strange though that none of the arguments of the function are then used. But I do not know if there is an alternative.

webmite, 2025-09-22:
The arguments of the function are used in the last if clause 
`

        if i > -1:
            # assign the selected reference
`
The self.refLines are the 4 references you can supply for alignment.  These are populated based on the accumulation of selections made.