FreeCAD/FreeCAD #22628: "Extend" tool in the Part Design workbench randomly changes position and angle of the line.

R2V1W3, 2025-07-20:
"extend" tool should only extend the selected line to the boundary and not change any other geometric elements of the line, such as angle, position, and location of the  the opposite end point regardless if the elements of the line are constrained or not in any way.

luzpaz, 2025-07-20:
Please add your full [About](https://wiki.freecad.org/About) info.  
Could you also add a screencast of what you are referring to for convenience of the testers and dev looking at this ticket ?

R2V1W3, 2025-07-21:
Select in any order, the line to be extended first, the boundary line second. 
The boundary line would change its angle while extending the first line. 

[image]

OS: macOS 12.7.6
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * MeshRemodel 1.10.37
  * boltsfc 2022.11.5

pixelwaster, 2025-07-21:
I went back and tried again with just the two lines. I _WAS ABLE_ to reproduce this. It also appears to matter _WHERE_ on the lines you click as well. I added a couple of parallel contraction likes to show the change better and one of the point constraint ended on the construction line. 

The behavior in the latest weekly (42605) is a little better because there is an indicator on the first clicked line that shows there the extension is going.  

In the last part of the picture, I moved the construction lines to the outside to avoid "interference"

[image]

ElPatoNoEsTu, 2025-07-21:
Sounds more like a Bug than a "feature" ...

R2V1W3, 2025-09-20:
I found today another unexpected behavior of the extend tool. If both lines are UNCONSTRAINED and Automatic constraint is OFF, the extend tool ignores the second selected line end just extends the first line to where the pointer is. 
So in order to extend the fist line to the limit line one needs to extend it past the limit line, then trim the extended line. And trm tool sets the constraint (evendough the Auto Constraint is OFF).

When Auto constraint is ON, the line still gets extended where the mouse pointer is when clicking, and moves the limit line (angle and endpoints) so they connect at that point, instead of what we said earlier, moving the end point along the direction of the line to touch the limit line. 

So as for now, extend and trim tools have radically different behavior, while most other CAD software have it as one tool, where if you click on the line to be extended/trimmed and then the limit object the tool will simply extend the line if it's short or trims the line if it's too long, constraining this connection in the process.