FreeCAD/FreeCAD #22629: Request for "close corner" new sketcher tool.

R2V1W3, 2025-07-20:
This is request for a new sketcher tool, "close corner", that would trim or extend one or both lines as required to result in coincident corner projected in extension of both lines. 
It would be equivalent to Fillet tool with default radius "0" and no creation of the fillet arc. 
It would replace consecutive use of trim extend tools combination with an important feature to preserve graphical elements of the original lines, such as angle and position, regardless of the constraint status of those lines.

R2V1W3, 2025-08-03:
When constructing either construction or normal geometry most often one needs to join two lines into coincident point (corner). Two intersecting lines can be too short, too long, or one short and one long. The new tool would extend or trim the lines as necessary while preserving angle and position of both lines, whether they are constrained or not. 

Most CAD software do this by allowing use of "filet tool with 0 radius", which closes the corner without creating the filet.  

It's by far the most used feature when constructing and editing sketches. 

It's equivalent to applying coincident constraint to end points of two lines, without changing position and angle of both lines, whether constrained or not.

PaddleStroke, 2025-08-03:
The fillet tool needs some rework as there are feature request for on-view-parameters. So it will change and be able to work as other CAD softwares.

PaddleStroke, 2025-08-03:
https://github.com/FreeCAD/FreeCAD/issues/14657 this is the issue that is about adding OVP to fillet. I'll link that here as they are going to be solved at the same time.

pierreporte, 2025-08-03:
This feature is really needed. `Sketcher_Extend` nearly does this but it only extends one side and never trims excess length. In other program the equivalent feature covers both extensions and the trim/extend requested here so it could be a good idea to expand `Sketcher_Extend`.

Note that it should work with all extendable geometry, not just lines. Restricted curves, that will be implemented by @AjinkyaDahale for https://github.com/FreeCAD/FPA-grant-proposals/issues/33 can be considered extendable while it remains smaller that the original.