FreeCAD/FreeCAD #22646: Illegible text color over dark background

Beep6581, 2025-07-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The solver messages link-label text color is a solid blue, and is almost impossible to read over a dark background:

[image]

Is it possible to change that somewhere?

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42644 (Git)
Build date: 2025/07/21 02:26:03
Build type: Release
Branch: main
Hash: 949767ea6925a86f138c3a92b631fb6c275e5548
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.38
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.13
  * Curves 0.6.64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-21:
cc @MisterMakerNL

Roy-043, 2025-07-21:
There are fine-tuning preferences for the colors of those texts:
https://wiki.freecad.org/Fine-tuning#Constraint_label_colors

AFAIU you need to specify a color name, for example "cyan".

Roy-043, 2025-07-21:
@MisterMakerNL I have commented on this before I think, but it is not good that the CFG files belonging to the themes do not all set the same parameters. Just from their file sizes you can see that this problem exists.

Syres916, 2025-08-29:
@MisterMakerNL taken from _FreeCAD.qss_ just now, these three entries should all use the exact same `@Variable` IMHO whether that be `@TextUrlColor` or something else as they all do the very same role:

```
Gui--UrlLabel {
  color : @TextUrlColor;
  text-decoration : underline;
}

QLabel[haslink="true"] {
  color: #71b6fb;
}

Gui--StatefulLabel[state="redundant_constraints"] {
  color : @SketcherRedundantConstraintsColor;
}
```
so this should be sufficient:

MisterMakerNL, 2025-08-29:
@Syres916 You are right there is TextUrlColor token.