FreeCAD/FreeCAD #22650: Selecting Body doesn't export complete body to 3mf

hansmerkl, 2025-07-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I select "Body" and export to 3mf, I don't get the holes. But when I select "Multitransform", the holes are exported. 

Files are attached

[3mf.zip](https://github.com/user-attachments/files/21352397/3mf.zip)
[FreezerHolder.FCStd.zip](https://github.com/user-attachments/files/21352396/FreezerHolder.FCStd.zip)

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * fasteners 0.5.38
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

Nothing else. Files are attached

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hansmerkl, 2025-07-21:
@luzpaz  Same result with 1.1

luzpaz, 2025-07-21:
@hansmerkl can you please add full [About](https://wiki.freecad.org/About) info? TIA

luzpaz, 2025-07-21:
* @hansmerkl also is it related to https://github.com/FreeCAD/FreeCAD/issues/22483

depthoffocus, 2025-09-05:
I don't believe this is a bug. Isn't it the expected behaviour?

In this model, the body's tip is set as the Groove before the MultiTransform. The Body exports at the tip.

[image]

If you set the MultiTransform as the tip, the Body will export with the holes.

[image]

closed 2025-11-05 by maxwxyz