FreeCAD/FreeCAD #22653: another possible regression in Sketcher

ElPatoNoEsTu, 2025-07-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Constraints can only be selected using the Mouse if they are "single" at the displayed place. 

[image] 

Like shown in the example -  a mouse over event at (1)  let the perpendicualr constrain get "blue"/selectable while trying the same at (2)  nothing happens whereas (3) behaves like (1) - a mouse over - let the constraint get "blue".

File:
[Test.zip](https://github.com/user-attachments/files/21354446/Test.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/111.979/1.25
```

### Subproject(s) affected?

None

### Anything else?

[Test.zip](https://github.com/user-attachments/files/21354446/Test.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-28:
Any diagnosis as to what triggered the regression ?

ElPatoNoEsTu, 2025-07-29:
I checked in: 

OS: Windows 10 build 19045
Architecture: x86_64
**Version: 1.1.0dev.41365 (Git)**
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.338/1.5
Installed mods: 
  * Curves 0.6.61
  * Defeaturing 1.2.2
  * DynamicData 2.70.0
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Manipulator 1.5.7

**The issue exists here too.**

ElPatoNoEsTu, 2025-07-29:
I tried also:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
**Version: 0.22.0dev.38314 (Git)**
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.61
  * Defeaturing 1.2.2
  * DynamicData 2.70.0
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Manipulator 1.5.7

**The Issue existed here too**.
Unfortuenately I did't kept much more older Builds to check there too.

closed 2025-08-25 by chennes