FreeCAD/FreeCAD #22654: Building issue (failed)

Dimitris75, 2025-07-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

missing from file ./src/Gui/InputHint.h

`#include <vector>`

required in line 303
 https://github.com/FreeCAD/FreeCAD/blob/5d58ac74599713341e6a30ef7a129cb4b5dce07f/src/Gui/InputHint.h#L303

### Full version info

```shell
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.42658 (Git)
Build date: 2025/07/21 21:06:49
Build type: Release
Branch: main
Hash: 5d58ac74599713341e6a30ef7a129cb4b5dce07f
Python 3.13.5, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7727/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kadet1090, 2025-07-22:
Interesting it build on CI and on my setup - can you elaborate on what platform and compiler you are using?

Dimitris75, 2025-07-22:
> Interesting it build on CI and on my setup - can you elaborate on what platform and compiler you are using?

Gcc on Debian 13. The compiler catched the error and tried to include vector at the beginning of the file, but, other errors arose from missing vector before that. 

By including vector the problem solved.

Dimitris75, 2025-07-22:
> Can you make a PR then? Would be much easier to merge - I won't be able to create PR until the end of the day

Do be honest, I've never made a PR before. I can try but, the and of your day it may be faster than me 😜