FreeCAD/FreeCAD #22663: Sketcher/Expressions. Incorrect path to sketch constraint when trying to print its name using expression.

kaiwas, 2025-07-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

My colleagues and I are solving a fairly simple problem.
We need to output the name of a constraint using an expression in text or a property.

[image]

The assigned name is output in a spreadsheet cell.

[image]

But it gives an error when trying to output it in a property.

[image]

It should be mentioned that in version 0.21 this algorithm works correctly

[image]

But if you open this file in 1dev, an error occurs when changing the name.

Test files can be downloaded from the link. (remove the zip extension)

[Const_name_1dev.FCStd.zip](https://github.com/user-attachments/files/21364830/Const_name_1dev.FCStd.zip)

[Const_name_0.21.FCStd.zip](https://github.com/user-attachments/files/21364835/Const_name_0.21.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42653 (Git) AppImage
Build date: 2025/07/21 16:35:39
Build type: Release
Branch: main
Hash: c99cd73cd60874f231bda6233477e9051afcea57
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Curves.backup1751366017.4487264 (Disabled)
  * addFC-main 1.6.2
  * Reinforcement
  * Curves.backup1751365993.7542331 0.6.63 (Disabled)
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Curves.backup1751366131.5540717 (Disabled)
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.64
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

Expressions

### Anything else?

Also checked in version 1.0.1 (does not work)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-07-22:
Addition.
It turned out that in 0.21 you can output having. But after that there is no way to change it.

kaiwas, 2025-10-17:
relevant

adrianinsaval, 2025-11-10:
The expression evaluates correctly in spreadsheets however the constraint cannot be renamed if used in a expression like this, but instead of keeping it's old name it actually looses it's name in 1.1dev (be93298). When a sketch is renamed all expressions that reference it are supposed to be updated to the new name, this includes expressions that reference it by number and that likely fails for a expression like this. It seems that expressions like this where not expected so one could argue that it having worked at all was a bug and we should stop this from being accepted in spreadsheet too

Note that a expression like `Sketch.Constraints.Nnname.Name` doesn't work on either case, I don't think it was ever intended to expose those properties of constraints in expressions

kaiwas, 2025-11-10:
@adrianinsaval Okay. Then there must be a way to get the constraint name outside the sketch.

mnesarco, 2026-01-12:
This is marked as a regression because it used to work prior to 1.0 (as far as I understand)
But I have a question, why `Sketch._self.Constraints[1].Name` works while `Sketch.Constraints[1].Name` don't? isn't it resolving to the exact same object?