FreeCAD/FreeCAD #22672: Stylesheets: "Requested non-existent design token" warnings in safe mode

furgo16, 2025-07-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When restarting in safe mode and selecting a theme via the Start page, the report window shows these warning messages:

```
Requested non-existent design token 'UserHighlight1Color'.
Requested non-existent design token 'UserHighlight1Color'.
Requested non-existent design token 'UserHighlight1Color'.
Requested non-existent design token 'UserHighlight1Color'.
Requested non-existent design token 'ToolbarButtonsPadding'.
Requested non-existent design token 'ToolbarButtonsPadding'
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42658 (Git) Snap 1588
Build date: 2025/07/21 21:06:49
Build type: Release
Branch: main
Hash: 5d58ac74599713341e6a30ef7a129cb4b5dce07f
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Assembly4 0.50.17
  * Road 2025.2.22
  * Launcher
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-25:
@MisterMakerNL is this related to the recent stylesheet PR you made ?

maxwxyz, 2025-07-26:
This is not only hapening in safe mode but anytime I open the app or change the theme on the start page.

closed 2025-10-20 by chennes