FreeCAD/FreeCAD #22680: Boolean Union works from object a -> object b but not the other way around

Excrubulent, 2025-07-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the attached file, I have created two objects from sketches that are driven from one another.

The FrontCase object is a basic plane with a curved edge.

The PalmCurve object was constructed with multiple sketch lines, the Surfaces -> Filling... tool, and the Part -> Loft tool, with the solid option set to true.  It was then used as the BaseFeature for a PartDesign object.

When I make FrontCase the active object, select the PalmCurve then use the PartDesign -> Boolean tool, the operation fails with these error messages:

`19:51:10  <Exception> TopoShapeExpansion.cpp(1396): Null shape`
`19:51:10  Boolean: Null shape`

In the boolean tool, if I switch to a Cut operation, the problem is the same.  If I switch to a Common operation, the are a few moment of delay, then the operation completes.

If I reverse the order of operations - I make PalmCurve active and select FrontCase - then the boolean operation completes without a problem, although doing subsequent operations may cause the problem to recur.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2319/1
```

### Subproject(s) affected?

PartDesign

### Anything else?

This is the thread where I first posted about the issue: https://forum.freecad.org/viewtopic.php?t=98543

File reproducing the issue, you will need to attempt the boolean operations in both directions to see the issue: [RetroPie_rebuild_test_04.zip](https://github.com/user-attachments/files/21385436/RetroPie_rebuild_test_04.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-10-17:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```

stefano2734, 2026-09-12:
It should be tested with occt 7.9.3 and 8.0.1 with actual versions 1.1.3 and 26.3 dev.