FreeCAD/FreeCAD #22681: CAM: chamfer bit errors in tool editor.

J-Dunn, 2025-07-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I wanted to use a chamfer bit and went to examine its shape and dimensions in the tool editors. 
It is full of errors which make it **unusable**. 

[image]

It calls itself a 45deg tool but angle alpha is not the cutting angle but the full sweep. 
In the table it says "cutting edge angle (alpha)", this contradicts the schematic.

In the table we read : "Diameter D" , yet in the schematic D is the tip diameter.  The full diameter is not shown at all. 

**The tool dimensions are over constrained.** 
Given full tool diameter D ;  cutting edge height h; and cutting angle , we can already determine the diameter of the truncated tip.  Thus specifying this dimension as well means the model is over constrained. 

Tip diameter is probably not a useful input, since hard to measure accurately. 

### Full version info

```shell
OS: Fedora Linux 40 (Forty) (LXQt/lxqt/xcb)
Architecture: x86_64
Version: 1.1.0dev.42540 (Git)
Build date: 2025/07/12 15:58:14
Build type: Release
Branch: DRFR
Hash: 566fe8371a88938e1b70a6048caac09f345cf024
Python 3.12.10, Qt 6.8.2, Coin 4.0.2, Vtk 9.2.6, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.0.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/98.4815/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

J-Dunn, 2025-07-23:
[image]

Opening an existing chamfer tool created this system with an earlier version of FreeCAD , angle alpha does not seem user accessible. 

That may not have been ideal but at least it worked.  So I'm marking this regression. 
Basically this is not in a usable state so a serious bug. 
I actually intended to test chamfer operations but without a configurable that is not possible. 

```
OS: Fedora Linux 40 (Forty) (LXQt/lxqt/xcb)
Architecture: x86_64
Version: 1.1.0dev.41585 (Git) AppImage
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/98.4815/1
```

J-Dunn, 2025-07-23:
What "needs confirmation" ?  
I posted screenshots from current master to show the bugs. 
This is not platform dependent and does not need reproducing.

J-Dunn, 2025-07-24:
A corollary of the incorrect diameter is that a drilling , sniffing for suitable tools, will select a chamfer bit with a dip diameter of 5mm (and an outer diam oc 12.33mm) as a candidate for drilling 5mm holes !  

That could be interesting to watch.

Connor9220, 2026-02-06:
@Connor9220 Look at this to see if this is a regression.

luzpaz, 2026-09-16:
@Connor9220 any chance you can confirm this ?

sliptonic, 2026-09-16:
> Obviously my not being interested is no reason to close a bug which has not been addressed. 

A bug that nobody is interested in seeing fixed isn't a bug.  It's a waste of time.
If even the reporter can't be bothered to update, test, report back, or otherwise improve the issue and nobody else is either, then close it and move on.  No shortage of things to spend our time on.

Connor9220, 2026-10-15:
**Reproduced:** No

**TL;DR**
- On current main (`38431477604`) the chamfer bit opens in the Tool Parameter Editor with no errors and can be edited.
- The overconstraint is fixed: the outer Diameter is now derived from tip diameter, cutting edge height and angle. It is read-only and no longer shown in the editor.
- The table now says "Tip diameter (D)", matching the drawing. α is the included angle in both the table and the drawing, and the operations use it the same way.
- Still there: the sample bit "45 Deg. Chamfer" has a 45° *included* angle (22.5° per side), not the 90° of a usual 45° chamfer mill. That is a naming issue in the sample data.

**What I tried**
1. Opened the shipped "45 Deg. Chamfer" in the Tool Parameter Editor. Fields: angle, height, length, shank, tip diameter (D). Derived Diameter is 10.26 mm, and the tool solid matches it.
2. Changed the angle to 90°. Diameter updated to 17.70 mm and the solid followed. No errors.
3. Loaded an old-style chamfer bit like the one in the second screenshot (height, Diameter, Length, Radius, Shank). It loads cleanly, and angle and tip diameter are now editable.

Fixed on main by "CAM: Make chamfer Diameter derived from tip and cutting geometry" (`48de20c8fd6`). The bearing-guided chamfer shape is a separate feature request.

[image]

**[Full repro page](https://connor9220.github.io/camprreviews/issues/22681/#review-1)**: what was tried, with screenshots

<sub>Automated campr run requested by @Connor9220 · tested on main `38431477604` · [repro page](https://connor9220.github.io/camprreviews/issues/22681/#review-1)</sub>

J-Dunn, 2026-10-15:
Thanks for tidying this up a bit. At least it is a bit more usable and consistent.
It seems there is still one major issue in using this for real tools.

> Tip diameter is probably not a useful input, since hard to measure accurately on the kind of tool shown in the model. It can be infered from the other dimensions.
> 
> The may be a need for another model similar to this, since may woodworking chamfer tools have a small ball bearing below the tip. This will have a precise dimension and also requires a further height dimension if future CAM WB is to include collision detection, where this would be crucial.

The tip is blunt and not a cutting edge. That diameter is not something which can be measured in practice and is not a dimension specified for the tools sold. The O/D is a machined edge and is usually physically measurable ( dependant on number of flutes). 

This is probably the dimension which the user needs to input when specifying the tool.

J-Dunn, 2026-10-16:
The truncated tip diameter (when there is no point) is not a dimension for these tools. The o/d is. 

https://vxb.com/products/120-degree-chamfer-milling-cutter-hss-end-mill-for
Few dimensions given but the last photo shows the shaft marked with 12.5x120deg HSS . Clearly o/d. even measuring the tip diameter as a means of accurately specifying the tool geometry is not possible. 

https://thetoolclub.co.uk/shop/chamfer-bit-w-brng-hw-s635-d45-x18/
Woodworking tools seem to typically have a bearing on the end. Measuring the end of truncated cone is physically impossible. This may need to be a different tool to include the bearing. 

https://www.amazon.com/chamfer-bits/s?k=chamfer+bits

The tools shape being modelled here does not really seem to match searches for "chamfer bit". Is there another name for this shaped tool which returns better matches?