FreeCAD/FreeCAD #22694: PartDesign: Additive pipe remove base geometry

Harvie, 2025-07-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am doing additive pipe of sketch along my part edge like this:

[image]

But when i click OK, only the "addition" is left, instead of it being added to original part:

[image]

### Full version info

```shell
This happens in FreeCAD 1.0.1, but also in latest weekly:

OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/07/21 18:24:28
Build type: Release
Branch: main
Hash: 80935791fd91ef89bfd3789fe1d6b76fbe0e384a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

PartDesign

### Anything else?

Here is the file, in case it helps:

[octavia3-dash-plate5.FCStd.zip](https://github.com/user-attachments/files/21395921/octavia3-dash-plate5.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-07-24:
Issue confirmed as described. An additive pipe around the edge would not produce a valid solid, fails check geometry. 

To check if a valid solid was possible, I tried part wb sweep. A valid fusion was produced with same sketch and sweep path (after converting to joinCurves). 

[image]

Version:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.64
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.24
  * Silk 0.2.0

maxwxyz, 2025-07-25:
@pieterhijma is this an OCC issue?