FreeCAD/FreeCAD #22695: Crashes after startup when any menu item relating to files is pressed

JaxonJJB, 2025-07-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As title says, FreeCAD crashes to desktop when I click things like "Parametric Part" under New File or go through the drop down menu File -> New. Any button that attempts to create a file in some way crashes it. I don't have any saved files to open to see if that also causes it. Attempting to open any Examples on the main landing also crash it. Top menu bar items and the "Open First Startup Menu" do not cause the issue. Persists in Safe Mode

It worked perfectly fine a few days ago (presumably v1.0?) and I ran a system update which updated FC to 1.0.1 and now this issue has shown up. Not urgent issue as I don't use FC much but figured I should still report it

Here is a pastebin of when I run FC from terminal: https://pastebin.com/UcmBkxfG

Running Arch Linux 6.15.7 w/ KDE Plasma 6.4.3 on AMD processor and integrated AMD GPU. 

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, OCC 7.9.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MasterChillerX, 2025-07-23:
I have the same problem, it seems FreeCAD has a problem with the new "Coin_4.0.4" package. I opened an issue [https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/23](url), but it seems to be no package problem.

ResRipper, 2025-07-24:
I also have the same issue, except the FreeCAD crashes right after the splash screen.

- OS: Arch Linux x86-64
- DE: KDE plasma + Wayland

<details><summary>Log</summary>
<p>

```
FreeCAD 1.0.1, Libs: 1.0.1R39285 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::124592411274176
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3def0) [0x715186e4def0]
#1  /usr/lib/libX11.so.6(XDefaultScreenOfDisplay+0x4) [0x715185ce6744]
#2  /usr/lib/libCoin.so.80(+0x461949) [0x715186861949]
#3  /usr/lib/libCoin.so.80(glxglue_init+0x43) [0x7151868627d3]
#4  /usr/lib/libCoin.so.80(cc_glglue_instance+0x826) [0x7151868635f6]
#5  0x7151866b800d in SoGLRenderActionP::isDirectRendering(SoState const*) const from /usr/lib/libCoin.so.80+0x4d
#6  0x7151866bd2c5 in SoGLRenderActionP::render(SoNode*) from /usr/lib/libCoin.so.80+0xc5
#7  0x7151866bcedc in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0xcc
#8  0x71518aabe33e in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x2e
#9  0x71518abdf87d in Gui::View3DInventorViewer::renderScene() from /usr/lib/freecad/lib/libFreeCADGui.so+0x17d
#10  0x71518ab8c655 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1d5
#11  0x71518ab9b6fa in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x4a
#12  0x71518875b3f3 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xab3
#13  0x7151887c7a76 in QFrame::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x36
#14  0x7151875675dc in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0xec
#15  0x715188701c60 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x80
#16  0x71518a6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#17  0x715187568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#18  0x71518874c2de in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib/libQt6Widgets.so.6+0x5e
#19  0x715188d829a3 in QOpenGLWidget::event(QEvent*) from /usr/lib/libQt6OpenGLWidgets.so.6+0x1a3
#20  0x715188701c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#21  0x71518a6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#22  0x715187568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#23  0x715188757c4d in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x1cd
#24  0x715188758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#25  0x7151887576c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x102
#26  0x715188757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#27  0x715188757955 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x395
#28  0x715188757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#29  0x715188758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#30  0x7151887576c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x102
#31  0x715188757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#32  0x715188758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#33  0x7151887576c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x102
#34  0x715188757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#35  0x715188758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#36  0x715188905563 in QMdiSubWindow::changeEvent(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x1f3
#37  0x71518875ab7f in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x23f
#38  0x715188701c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#39  0x71518a6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#40  0x715187568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#41  0x715188748fb7 in QWidget::setWindowState(QFlags<Qt::WindowState>) from /usr/lib/libQt6Widgets.so.6+0xb7
#42  0x71518874923b in QWidget::showMaximized() from /usr/lib/libQt6Widgets.so.6+0x2b
#43  0x71518890102c in QMdiSubWindow::eventFilter(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x88c
#44  0x7151875675dc in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0xec
#45  0x715188701c60 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x80
#46  0x71518a6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#47  0x715187568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#48  0x715188748fb7 in QWidget::setWindowState(QFlags<Qt::WindowState>) from /usr/lib/libQt6Widgets.so.6+0xb7
[... cut ...]
#77  0x71518783c5e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x72
#78  0x7151875744b6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x206
#79  0x71518756c7c1 in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0xb1
#80  0x71518a5c7bf5 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x9b5
#81  freecad(+0x8276) [0x635e2a2ac276]
#82  /usr/lib/libc.so.6(+0x276b5) [0x715186e376b5]
#83  /usr/lib/libc.so.6(__libc_start_main+0x89) [0x715186e37769]
```

</p>
</details>

dlanyar, 2025-07-24:
Same issue here, when in click on any button the app crashes.

You'll find attached error messages that i got from the terminal.

https://pastebin.com/BzPhUpmv

 ```bash
OS: GNU/Linux (KDE/wayland)
Architecture: x86_64
FreeCAD version: 1.0.1-5
Python: Python 3.13.5
Qt: 6.9.1
Coin: coin 4.0.4-1
VTK: vtk 9.5.0-4
OCC: opencascade 1:7.9.1-1
Locale: LANG=en_US.UTF-8

Nicuz, 2025-07-24:
Same issue for me on CachyOS, FreeCAD crashes after startup when clicking on any button. Temporarily downgraded to Coin 4.0.3 as @MasterChillerX suggested and it works.

* **OS**: CachyOS x86_64
* **WM**: Hyprland 0.50.1 (Wayland)
* **FreeCAD**: 1.0.1.39285 (Git)

JaxonJJB, 2025-07-24:
Can also confirm downgrading the coin package 4.0.3 works, thanks for the tip

BellusMaxima, 2025-07-24:
Same issue here, running CachyOS with AMD GPU and iGPU.

I also found that running `sudo freecad` in console also solved the crashing, but I have no clue why.

luzpaz, 2025-07-24:
Has anyone reported the coin issue upstream yet ?

Syres916, 2025-07-25:
> Has anyone reported the coin issue upstream yet ?

The Maintainer is well aware see https://github.com/coin3d/coin/issues/548#issuecomment-3042499779

xyproto, 2025-07-28:
Coin was also downgraded to 4.0.3 on Arch Linux as well, for now: https://archlinux.org/packages/extra/x86_64/coin/

lonyelon, 2025-07-28:
> Coin on my system is already 4.0.3, so downgrading will not help me.
> […](#)
> 
>  <[#22695 (comment)](https://github.com/FreeCAD/FreeCAD/issues/22695#issuecomment-3126026666)>
> 
>  Coin was also downgraded to 4.0.3 on Arch Linux as well, for now:
>  https://archlinux.org/packages/extra/x86_64/coin/
> 
>  <[#22695 (comment)](https://github.com/FreeCAD/FreeCAD/issues/22695#issuecomment-3126026666)>,

I have not been able to reproduce the issue with coin 4.0.3.

luzpaz, 2026-09-20:
@tritao is it worth patching Coin, if this is a valid workaround ?

tritao, 2026-09-20:
I think we should patch Coin to work out of the box, but there was a recent PR that got merged https://github.com/FreeCAD/coin/pull/12 that changes EGL code a lot.

It will be part of next weekly build so please check if it still is an issue then.