FreeCAD/FreeCAD #22709: Sketcher: Minimum and maximum length and angle constraints

UnKabaraQuiDev, 2025-07-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Minimum and maximum constraints would be very useful in the Sketcher wb, they would allow users to define ranges for dimensions instead of locking a single value. This is especially useful when building variant-based models or adaptive parts, where some dimensions must stay within some limits, but don't need to be fixed.

This would also help with issues like [#21697](https://github.com/FreeCAD/FreeCAD/issues/21697), where we need the solver to converge to a single specific solution. (In that case, two circles admit two valid intersection points, but we want to restrict the solution to only one of them. A minimum constraint can guide the solver toward the correct configuration, without overconstraining the sketch.)

There's also an open proposal to support inequality constraints more generally [#18700](https://github.com/FreeCAD/FreeCAD/issues/18700), which includes min, max, and preferred values. That’s a broader feature and likely more complex to implement.

I have no idea how to implement this (I not very good at c++) feedback is greatly appreciated

### Full version info

```shell
(not needed)

OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FreecadDiscordPresence 1.0.2
  * Curves 0.6.64
  * Behave-Dark-Colors 0.1.1
  * A2plus 0.4.68
  * 3DfindIT 1.2.0
  * Reinforcement
  * FreeCAD-themes 2025.6.4
  * fasteners 0.5.39
  * Assembly4 0.60.1
  * lattice2 1.0.0
  * freecad.gears 1.3.0
  * 3D_Printing_Tools
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-07-24:
I think I can do such thing. Remind me by the end of the year if nothing is coming.

kaiwas, 2025-07-24:
> I think I can do such thing. Remind me by the end of the year if nothing is coming.

That would be great! I suppose that with the help of such constraints it would be possible to fix flipping sketches.

UnKabaraQuiDev, 2025-12-07:
I'll close this issue , hope #24242 gets merged soon !

closed 2025-12-07 by UnKabaraQuiDev