FreeCAD/FreeCAD #22716: Draft: snapping_extention doesn´t work anymore in Vers.1.01

ChristianKaspar, 2025-07-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The current version 1.01 cannot catch an extension of a line on a circle. I had increased the node selection radius to 30p. Also, snapping to the intersection of two lines converging towards each other does not work. Extending along a line is also not possible. The line is preselected and the end point is marked, but an extension in the direction of the line, i.e. the cursor movement, is not displayed. In my opinion, however, this already worked in version 1.0.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCad_MyDark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4
  * BillOfMaterials 1.0.3.3
  * CurvedShapes 1.0.13
  * Curves 0.6.64
  * dodo 1.0.1
  * EasyProfileFrame 0.0.1
  * fasteners 0.5.39
  * FrameForge 0.1.3
  * freecad.gears 1.3.0
  * ImportNURBS 1.1.0
  * Manipulator 1.6.0
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.7.24
  * ShortCuts
```

### Subproject(s) affected?

None

### Anything else?

Hope, problem can fixed soon.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-07-24:
It seems to work as I would expect using the Linux AppImage, have you tried Help menu > Restart in Safe Mode (only change the navigation style to what you like and test it as a new user?

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark_Pre1_0.qss/Behave-dark_Pre1_0/Qt default
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * Silk 0.2.0
  * sheetmetal 0.7.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.31.4
  * ChangeAppFont
  * Quetzal 1.5.6
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * FreeCAD-Reporting.zip
  * Curves 0.6.64
  * QetWireManager
  * FreeCAD-Reporting
  * AddonManager 2025.7.17
  * A2plus 0.4.68

```

ChristianKaspar, 2025-07-24:
You are right. It seems to be a performance problem of my computer. On my 2nd computer, snapping works.

closed 2025-07-25 by ChristianKaspar