FreeCAD/FreeCAD #22718: Do not use capitalization in path names

Beep6581, 2025-07-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While researching the stylesheets issue #22717 I noticed that the given path is `Gui/Stylesheets/`
I tried `~/.local/share/FreeCAD/Gui/Stylesheets/foo1.qss` and it worked.
Then I tried to use an all-lowercase path of `~/.local/share/FreeCAD/gui/stylesheets/foo2.qss` but it was not picked up.

Paths should not be capitalized. If for some reason you want to capitalize them, then at least capitalize them correctly: "GUI" not "Gui".

As for "style sheets", it is used inconsistently in FreeCAD as both one word and as two:
grep -rni "style sheet" programs/code-freecad 2>/dev/null | wc -l
225
grep -rni "stylesheet" programs/code-freecad 2>/dev/null | wc -l
838

Qt also uses it inconsistently:
https://doc.qt.io/qt-6/stylesheet-reference.html

You avoid the `/StyleSheets` vs `/Stylesheets` problem by not capitalizing it at all, `/stylesheets`

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42687 (Git)
Build date: 2025/07/24 14:59:58
Build type: Release
Branch: main
Hash: cdd69cca7dcdd6924bf84b7646c641043afce293
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.39
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.13
  * Curves 0.6.64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-25:
@FreeCAD/code-quality-working-group

hyarion, 2025-07-25:
We use PascalCase a lot, and it is common to change acronyms then as well.
`GuiController` instead of `GUIController` as it makes it easier to read. That said, making the paths case-insensitive might be a good option

Beep6581, 2025-07-26:
PascalCase - sure, for code, but not for paths. It's just easier to lowercase everything for paths, and doing so avoids cross-OS issues. Same as not using characters outside of the ASCII set in code or in paths and filenames.
e.g. in `tools/build/WindowsInstaller/lang` you use lowercase English names with characters within the ASCII set for languages, e.g. `slovak.nsh`, instead of `Slovenčina.nsh`, `japanese.nsh` instead of `日本語.nsh`, etc.
You'd avoid situations where code was written by someone using Windows where the OS is case-insensitive and it ends up not working when run in Linux or macOS.

Further random example of inconsistency:
`~/.config/FreeCAD/CamAssets/Library/Default.fctl` (capitalized)
vs
`~/.config/FreeCAD/CamAssets/Shape/example.fcstd` (lowercase)

I think it would simplify things for FreeCAD developers as well as for users and package maintainers if path and file names were consistently lowercase.

hyarion, 2025-07-26:
There is a readability issue with lowercase if you concatenate multiple words asitgetsdifficulttoreadwithoutanytypeofseparation. For instance compare `CamAssets` vs `CAMAssets` vs `camassets`, personally I can read `CamAssets` way easier than the others and I'm sure there are other, better cases that we currently use.

kadet1090, 2025-07-26:
That would be a huge change - for a little to no gain. For now our file names are mostly aligned with classes that they define. Using capitalization in filenames is common in most programming languages (PHP, Java, C# - as examples) and in general not considered as problem. I don't think that here is an action to take here apart from being more cautious about ensuring that filenames are passed with correct casing.

Beep6581, 2025-08-14:
I agree that attempting to implement consistency at this point would be a huge effort that would lead to bugs and not be worth the effort. However, my intention with this issue is for FreeCAD to at least implement and express some capitalization standard *going forward*, so that with time you end up with consistency without needing a mega-patch.