FreeCAD/FreeCAD #22721: Boolean cut fails in Part Design workbench

paranoidi, 2025-07-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am trying to cut outer threads to a part with part design boolean operation (cut) but no matter what I try it will fail. Result rotates the part  randomly and complains that "Boolean: Result has multiple solids: that is not currently supported.
".  This should not result multiple solids in my uneducated analysis. I did try resizing and moving the cutter extensively and it made no difference. Turning off often problematic refine made no difference either.

The cut works correctly using Part workbench boolean cut. It should work in Part Design workbench too. I was able to cut the inner threads for counterpart fine (not included in the example file).

I have attached the example project file in it's simplest form. It has just one revolve and fastener to do the boolean cut.

[cut_fails.FCStd.zip](https://github.com/user-attachments/files/21418465/cut_fails.FCStd.zip)

### Full version info

```shell
FreeCAD 1.0.1 version fe624f2c3ab8af3c1c01ddcaaebe82091e217588
Tested also with daily version fe624f2c3ab8af3c1c01ddcaaebe82091e217588
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paranoidi, 2025-07-25:
Tested again today with daily:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42705 (Git)
Build date: 2025/07/25 02:00:45
Build type: Release
Branch: main
Hash: 705f2daa61b2cab2a3cb4885ac206bb775232af5
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * fasteners.backup1746132246.8582077 (Disabled)
  * fasteners
  * fasterners 0.5.38
```

Let me know if there's anything else I can do to help.

macdroid53, 2025-11-29:
Not sure if it is part of the same issue, but:

[image]

If you create a solid (a Part WB cube, from Fasteners workbench, etc.) add it as a Basefeature of a new Body. The attempt to use PD Boolean Cut between the two. It sucks the source of the Basefeature,  as well as the the Body/Basefeature. (And in the case of a Part Cube, it duplicates the cube as well. FOr a fastener, it just sucks in the fastener.

This was originally seen in 1.02 and is shown here in 1.2dev.

[BogusPDBoolean.FCStd.zip](https://github.com/user-attachments/files/23834846/BogusPDBoolean.FCStd.zip)

macdroid53, 2025-12-01:
The original find of the duplicate in the tree actually added the fastener as shown above. But it also did some strange relocation of the original (non-PD) fastener. And, it appeared to use both the re-positioned solid and the Basefeature version in the Boolean Cut, resulting in overlap and coplanar issue in the final result.

I any case, I don't think the original non PD object should end up in the tree when the Boolean operation is to be between the Body-w-Basefeature and the active Body.

If you attempt to Boolean Cut between another Body and Body-w_Basefeature, the preview looks like this:

[image]