FreeCAD/FreeCAD #22722: FCInfo.png is missing.

leoheck, 2025-07-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am seeing this issue lately, is this a bug? How can I fix this?

17:52:20  Cannot find icon: /home/lheck/.cache/FreeCAD/Cache/AddonManager/MacroIcons/FCInfo.png

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2025-07-25:
A couple of questions for you: first, do you have the FCInfo macro installed? Second, do you have a button for it installed into a toolbar? Third, is the file FCInfo.png in your user macros directory?

maxwxyz, 2025-07-25:
Is this an issue of the macro itself?

leoheck, 2025-07-25:
Funny, I removed the macro with the AddonManager, restarted FC, and I am still seeing this

```
15:25:58  Cannot find icon: /home/lheck/.cache/FreeCAD/Cache/AddonManager/MacroIcons/FCInfo.png
``` 

Let me check if the Safe Mode.

leoheck, 2025-07-25:
By the way, this is my toolbar (Addonmanager and CarbonCopy)
It has 2 extra icons at the end of the first row.

 [image]

leoheck, 2025-07-25:
Weird that neither of them is in the toolbar.
There is no FCInfo, and no Section (that was missing in the previous weekly builds)

leoheck, 2025-07-25:
Ok, fixed this thing by removing this "Auto-Created Macro Toolbar"
I did not do this. 

Now, who auto-created this?

closed 2025-07-25 by leoheck