FreeCAD/FreeCAD #22726: QApplication->topLevelWidgets() appears to return a list with null or that can change.

schmidtw, 2025-07-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Running against fairly up to date code, I experienced a crash due to what appears to be a null pointer dereference.

Looking at the stack trace it appears that in the code: https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Gui/Action.cpp#L618 the value of `widget` is either `null` or the encountered widget is being destroyed.

My very limited understanding of QT's gui code is that it's single threaded, but inside calls to QT it could be re-entrant or do other things to the overall state of the object model.  If this is the case, then where `QApplication::topLevelWidgets();` is called & a list is iterated over, the iteration process must check if the item is `null` for each item if any QT calls are ever made between when the list is acquired and iterated upon.  (I could be completely wrong about this...)

Unfortunately, I've not been able to reproduce this issue.

I propose adding a null check like this:
```cpp
void ActionGroup::onHovered(QAction *act)
{
    const auto topLevelWidgets = QApplication::topLevelWidgets();
    QMenu* foundMenu = nullptr;

    for (QWidget* widget : topLevelWidgets) {
        if (!widget || !widget->isVisible()) {  // New checks here
            continue;
        }
...
```

Here are the other places I see that have similar patterns/potential for issues:
- https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Gui/
- https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Gui/MainWindow.cpp#L1759
- https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Mod/Plot/Plot.py#L56
- https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Mod/TemplatePyMod/Automation.py#L21
- https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Mod/TemplatePyMod/TaskPanel.py#L85
- https://github.com/FreeCAD/AddonManager/blob/34d433a02c7ec5c73bec9c57d0a27ea70b36c90d/addonmanager_utilities.py#L549
- https://github.com/FreeCAD/FreeCAD/blob/705f2daa61b2cab2a3cb4885ac206bb775232af5/src/Tools/embedded/Qt/mainwindow.cpp#L152

## Crash log
```
Sheet Metal workbench loaded
Ignore recovery file  "/home/schmidtw/.cache/FreeCAD/Cache/FreeCAD_Doc_dabd5654-8274-4479-a834-f9a8682627ff_7de22b_22924/fc_recovery_file.fcstd"  because it is older than the project file "/home/schmidtw/projects/sv08-psu-upgrade/Combined.FCStd" 

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x19c30) [0x7f7873227c30]
#1  0x7f787215e408 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x18
#2  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#3  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#4  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#5  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#6  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#7  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#8  0x7f787215e8a1 in qt_qFindChildren_helper(QObject const*, QAnyStringView, QMetaObject const&, QList<void*>*, QFlags<Qt::FindChildOption>) from /lib64/libQt6Core.so.6+0x4b1
#9  0x7f7877391c63 in QList<QMenu*> QObject::findChildren<QMenu*>(QAnyStringView, QFlags<Qt::FindChildOption>) const from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x5b
#10  0x7f787738fd60 in QList<QMenu*> QObject::findChildren<QMenu*>(QFlags<Qt::FindChildOption>) const from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x40
#11  0x7f787738588f in Gui::ActionGroup::onHovered(QAction*) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0xaf
#12  0x7f7877397732 in QtPrivate::FunctorCall<std::integer_sequence<unsigned long, 0ul>, QtPrivate::List<QAction*>, void, void (Gui::ActionGroup::*)(QAction*)>::call(void (Gui::ActionGroup::*)(QAction*), Gui::ActionGroup*, void**)::{lambda()#1}::operator()() const from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x70
#13  0x7f787739879d in void QtPrivate::FunctorCallBase::call_internal<void, QtPrivate::FunctorCall<std::integer_sequence<unsigned long, 0ul>, QtPrivate::List<QAction*>, void, void (Gui::ActionGroup::*)(QAction*)>::call(void (Gui::ActionGroup::*)(QAction*), Gui::ActionGroup*, void**)::{lambda()#1}>(void**, QtPrivate::FunctorCall<std::integer_sequence<unsigned long, 0ul>, QtPrivate::List<QAction*>, void, void (Gui::ActionGroup::*)(QAction*)>::call(void (Gui::ActionGroup::*)(QAction*), Gui::ActionGroup*, void**)::{lambda()#1}&&) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x24
#14  0x7f7877397795 in QtPrivate::FunctorCall<std::integer_sequence<unsigned long, 0ul>, QtPrivate::List<QAction*>, void, void (Gui::ActionGroup::*)(QAction*)>::call(void (Gui::ActionGroup::*)(QAction*), Gui::ActionGroup*, void**) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x60
#15  0x7f7877395ef2 in void QtPrivate::FunctionPointer<void (Gui::ActionGroup::*)(QAction*)>::call<QtPrivate::List<QAction*>, void>(void (Gui::ActionGroup::*)(QAction*), Gui::ActionGroup*, void**) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x44
#16  /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so(+0x394667) [0x7f7877394667]
#17  /lib64/libQt6Core.so.6(+0x1657ba) [0x7f78721657ba]
#18  0x7f7872d9543a in QActionGroup::hovered(QAction*) from /lib64/libQt6Gui.so.6+0x4a
#19  /lib64/libQt6Core.so.6(+0x1657ba) [0x7f78721657ba]
#20  0x7f78736c6158 in QToolButton::enterEvent(QEnterEvent*) from /lib64/libQt6Widgets.so.6+0x38
#21  0x7f78734a6908 in QWidget::event(QEvent*) from /lib64/libQt6Widgets.so.6+0x2b8
#22  0x7f787343d97f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x8f
#23  0x7f787730fa79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x133
#24  0x7f78720f9f38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x168
#25  /lib64/libQt6Widgets.so.6(+0x45c2d) [0x7f7873445c2d]
#26  0x7f7873446d53 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib64/libQt6Widgets.so.6+0x523
[... 53 lines cut ...]
#80  0x7f7872103419 in QCoreApplication::exec() from /lib64/libQt6Core.so.6+0xa9
#81  /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so(+0x1571f0) [0x7f78771571f0]
#82  /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so(+0x1574ce) [0x7f78771574ce]
#83  0x7f787715787b in Gui::Application::runApplication() from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x2b1
#84  ./bin/FreeCAD() [0x401cc1]
#85  /lib64/libc.so.6(+0x35f5) [0x7f78732115f5]
#86  /lib64/libc.so.6(__libc_start_main+0x88) [0x7f78732116a8]
#87  ./bin/FreeCAD() [0x4009a5]
```

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR42678 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-25:
@kadet1090 FYI

chennes, 2025-07-26:
Qt's example code doesn't show any hint that this might be possible (and I don't think widget can actually be `nullptr`, even if it's been invalidated, so that check probably doesn't do anything)": https://doc.qt.io/qt-6/qapplication.html#topLevelWidgets
```
void showAllHiddenTopLevelWidgets()
{
    const QWidgetList topLevelWidgets = QApplication::topLevelWidgets();
    for (QWidget *widget : topLevelWidgets) {
        if (widget->isHidden())
            widget->show();
    }
}
```
All of the Qt widgets should exist only in the main thread, and I think their creation will actually crash if you try to do it on another thread, so I'm not sure that's what's going on here. If we can create a test of some kind that reproduces the issue more reliably then we'd have some thing to go on. Probably needs to be a fuzzer of some sort.

schmidtw, 2025-07-26:
That's reasonable.  I've tried to replicate it several times with no luck.  I've locally added the null check & print out a debug statement in the hopes that I can catch it while generally using the dev code.

schmidtw, 2025-10-17:
I've not seen this specific example again.  I'm not convinced it's not still lurking in the code, but I this probably can be closed.  If we see it again, we can always re-open it.

luzpaz, 2026-01-16:
> I've not seen this specific example again. I'm not convinced it's not still lurking in the code, but I this probably can be closed. If we see it again, we can always re-open it.

Closing.