FreeCAD/FreeCAD #22733: Crash on align to selection

maxwxyz, 2025-07-25:
I've noticed a hard crash on one case: 
`Unhandled exception`
File: 
[AlignCrash.FCStd.zip](https://github.com/user-attachments/files/21372798/AlignCrash.FCStd.zip)

When trying Std_AlignToSelection on such corner faces.

[image]

_Originally posted by @maxwxyz in https://github.com/FreeCAD/FreeCAD/issues/22365#issuecomment-3104095175_

maxwxyz, 2025-07-25:
@Rexbas I've created an issue to track this

chennes, 2025-07-26:
The failure is really on line 1806 of PartFeature.cpp:
```
            if (!edgeTopoShape.isLinearEdge()) continue;
```

maxwxyz, 2025-07-26:
I think in this case the camera view should still be oriented to the face normal but not necessarily rotated/aligned to align an edge to the view.

closed 2025-07-26 by kadet1090

chennes, 2025-07-26:
> @chennes your PR just is a workaround for the crash but does not fix the view alignment, correct?

It makes the view alignment work as intended by eliminating the crash 🙂. There is no way to choose an up or down direction for those corners, so the alignment is only making the face's representative normal point at (or away from) the camera.

maxwxyz, 2025-07-26:
thanks, I was just curious if there needs to be an additional change in the alignToSelection

PR #22745 (2025-07-26): Part: Add error handling to TopoShape::isLinearEdge
Fix #22733 by making sure that the resolved `Geom_Curve` is valid before checking if it is linear. There are several alternative places where we could be checking this, and/or handling the exception that is raised if we fail to check this. I picked what seemed to be the least intrusive option.