FreeCAD/FreeCAD #22734: PartDesign: Unnecessary edge geometry after pocketing curved surface?

Harvie, 2025-07-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,
i have curved surface like this and i've made pocket sketch to make it round.

[image]

After cuting the pocket, it seems the edges are unnecessarily split into two pieces (one long and one short)

[image]

Those seem to be tangent, so i think it should be single edge....

### Full version info

```shell
(tested in 1.0.1 release as well)

OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/07/21 18:24:28
Build type: Release
Branch: main
Hash: 80935791fd91ef89bfd3789fe1d6b76fbe0e384a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

PartDesign

### Anything else?

This seems to trigger more bugs, down the line, eg: #22694

Here is the file: 

[extra-geometry-bug.FCStd.zip](https://github.com/user-attachments/files/21430159/extra-geometry-bug.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-12:
@tomate44 any chance you can weigh in as well ?

FEA-eng, 2025-10-17:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```