FreeCAD/FreeCAD #22747: Stylesheet: toolbar buttons reduced hover and activation visual feedback

furgo16, 2025-07-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Recent changes to the stylesheets (perhaps) have resulted in reduced visual feedback when interacting with toolbar buttons:

- On hover the frame around the current toolbar button is no longer visible
- On click, the darkened background on the current toolbar button is no longer shown

Behavior in 1.0:

https://github.com/user-attachments/assets/d77ff7e3-f24d-4613-a590-8d77ed112f30

Current behavior in 1.1.0-dev

https://github.com/user-attachments/assets/2ff1a162-07df-4118-880a-fdea0141cee3

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42727 (Git) Snap 1598
Build date: 2025/07/25 22:54:43
Build type: Release
Branch: main
Hash: ce9bc869086159a8655e262500f00bd39cb4d860
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Launcher
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-26:
cc @MisterMakerNL

MisterMakerNL, 2025-07-26:
I try not to be the designer just the maintainer although abit true the ddw soo  @FreeCAD/design-working-group

furgo16, 2025-08-13:
@FreeCAD/design-working-group what are your thoughts on this one? To me, it's a significant regression, but others might see it differently.