FreeCAD/FreeCAD #22758: Sketcher: Sketcher_Switch_Virtual_Space causing crash

ErtugrulEge, 2025-07-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!

Steps to reproduce:
1.Create a part design file and body
2.Create a sketch and draw a simple rectangle
3.Select Constraint1 (or another one)
4.Press, `Sketcher_Switch_Virtual_Space `
5.Now, press an empty place on the screen
6.Press several times to `Sketcher_Switch_Virtual_Space `

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-07-27:
Crash confirmed. Press two times on that virtual space button, and the software crashes.

[image]

Version:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.64
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.24
  * Silk 0.2.0

maxwxyz, 2025-07-27:
@AjinkyaDahale FYI

maxwxyz, 2025-07-27:
It crashed on windows even without a selection

matthiasdanner, 2025-07-27:
It has Most Probably to do with #22072. i'll have a Look next week

matthiasdanner, 2025-07-27:
The problem here is, that with this button the user can show and hide constraints the user selected. There is a option to only display the selected constraints, which is conflicting with the user being able to select the displayed constraints.

I created a PR to revert the changes for making the option to only display selected constraints work

PR #22780 (2025-08-09): Sketcher: Fix crash on set constraints to virtual space

## Issues
Fixes #22758 and #22769
Reverts the fix for #9373 from #22072

The Option for only displaying constraints which are filtered in the constraint list and the button to switch some or all displayed constraints pursue conflicting goals. On the one hand you have some rules which constraints are displayed (the filtered list) and on the other hand the user can choose which constraints are displayed.

I reverted the fix for the filtered constraints to mitigate the crash.
In my opinion the option for the filtered constraints has to be r [... cut]
closed 2025-08-09 by chennes