FreeCAD/FreeCAD #22760: FEM Workbench: exported xdmf files contain wrong TopologyType for second order meshes

violettinged, 2025-07-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

For second order meshes created using gmsh in the FEM workbench, the `TopologyType` of 10-node-tetrahedrons and 6-node-triangles in the xdmf output are unknown to FEniCSx. They should be changed to `Tetrahedron_10` and `Triangle_6`, see:

https://www.xdmf.org/index.php/XDMF_Model_and_Format.html#XML_Element_(Xdmf_ClassName)_and_Default_XML_Attributes

### Full version info

```shell
OS: macOS 14.7.4
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

FEM

### Anything else?

The issue was already raised here: https://forum.freecad.org/viewtopic.php?style=4&p=328456#p328456

As a workaround, I currently run a script on the xdmf-file generated by FreeCAD and replace `Tet_10` with `Tetrahedron_10` and `Tri_6` with `Triangle_6`. This works for me in the newest Version of FEniCSx (dolfinx0.9.0).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-27:
Was this tested on 1.1dev ?  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

violettinged, 2025-07-27:
Tested on Version: 1.1.0dev.42721 (Git). Problem persists.

luzpaz, 2025-07-28:
Full About info pls...TIA!

cc @FEA-eng

luzpaz, 2025-08-14:
@FEA-eng can you confirm ?

luzpaz, 2025-08-16:
@violettinged any chance you can test and confirm the fix ?

violettinged, 2025-08-20:
I have tested it and can confirm that the issue is solved.

PR #23141 (2025-08-25): FEM: Fix xdmf export of second-order meshes
fixes #22760

@violettinged FYI
closed 2025-08-25 by chennes