FreeCAD/FreeCAD #22762: FEM: xdmf mesh export fails for vertex/node mesh groups

violettinged, 2025-07-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Create a cube
2. Switch to the FEM Workbench
3. Select the cube and create `FEM mesh from shape by Gmsh`
4. Create `FEM mesh group` by selecting a corner vertex of the cube
5. Start the meshing process
6. Export the mesh by selecting `FEM mesh Fenics (*.xml *.xdmf)` and specifying the file-extension `.xdmf`

This procedure gives different results:

With the `FEM Gmsh Mesh Params` option `Groups of Nodes` set to `false`, a xdmf-file is exported but no mesh groups are detected and the xdmf-file does not contain a mesh group for the vertex.

With the `FEM Gmsh Mesh Params` option `Groups of Nodes` set to `true` the dialog `Mesh groups detected. Please choose values for the different groups` does show up, but the export fails with an error message.	

### Full version info

```shell
OS: macOS 14.7.4
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

violettinged, 2025-07-27:
Tested on Version: 1.1.0dev.42721 (Git). Problem persists.

luzpaz, 2025-07-28:
@violettinged can you please add the full [About](https://wki.freecad.org/About) info of the 1.1dev version you tested on? TIA!

luzpaz, 2025-07-28:
cc @FEA-eng

violettinged, 2025-07-28:
OS: macOS 14.7.4
Architecture: arm64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2

violettinged, 2025-07-29:
Mesh groups are also exported to .xdmf without any further configuration needed. It works perfectly fine for volumes, faces and edges.

FEA-eng, 2025-07-29:
When I try to export a mesh with mesh group on a face to .xdmf, I click Ok in the dialog but then there's an error:

```
19:03:35  pyException: Traceback (most recent call last):
  File "D:\FreeCAD_weekly-builds-42780-conda-Windows-x86_64-py311\Mod\Fem\feminout\importFenicsMesh.py", line 131, in accept
    writeFenicsXDMF.write_fenics_mesh_xdmf(
  File "D:\FreeCAD_weekly-builds-42780-conda-Windows-x86_64-py311\Mod\Fem\feminout\writeFenicsXDMF.py", line 270, in write_fenics_mesh_xdmf
    mesh_function_codim = dim_cell - FreeCAD_Group_Dimensions[mesh_function_type]
                                     ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
<class 'KeyError'>: ('Node',)
```

violettinged, 2025-07-29:
> When I try to export a mesh with mesh group on a face to .xdmf, I click Ok in the dialog but then there's an error:
>
> ```
> 19:03:35  pyException: Traceback (most recent call last):
>   File [...]

This is because it tries but fails to export the vertices of the face mesh group. Uncheck `Groups of Nodes` and this error should disappear.

FEA-eng, 2025-07-29:
> Uncheck `Groups of Nodes` and this error should disappear.

Right. I thought I had it disabled at first when this error occured but I can't reproduce it now without that property enabled.

Exporting to .xml (for Fenics) won't include mesh groups, right? I'm not familiar with those formats and that solver.

violettinged, 2025-07-30:
The currently recommended mesh format for FEniCSx (which is the new and actively developed version of the legacy FEniCS library) is the .xdmf format. The old `dolfin xml` format has been deprecated for years even in legacy FEniCS. I am not quite sure whether it is possible to include mesh groups (so called mesh tags in FEniCS) into a single `dolfin xml`mesh file. But even if it was, implementing this functionality is not worth the effort anymore.

FEA-eng, 2026-05-24:
The `Groups of Nodes` property was removed in #29440. Now there's just this error:

```
10:41:42  pyException: Traceback (most recent call last):
  File "D:\FreeCAD_weekly-2026.05.20-Windows-x86_64\Mod\Fem\feminout\importFenicsMesh.py", line 131, in accept
    writeFenicsXDMF.write_fenics_mesh_xdmf(
  File "D:\FreeCAD_weekly-2026.05.20-Windows-x86_64\Mod\Fem\feminout\writeFenicsXDMF.py", line 269, in write_fenics_mesh_xdmf
    mesh_function_codim = dim_cell - FreeCAD_Group_Dimensions[mesh_function_type]
                                     ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
<class 'KeyError'>: ('0DElement',)
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260519 (Git shallow)
Build date: 2026/05/19 20:59:45
Build type: Release
Branch: (HEAD detached at 5a5d8d042)
Hash: 5a5d8d0425fdb440fb5dcb47f1c48422d07d3267
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

maxwxyz, 2026-11-07:
There is a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/edd866fb209a220fa043a634e78249a1db8c2038

It adds node groups to the XDMF writer instead of treating the node IDs as element IDs. Could be picked up for a PR.