FreeCAD/FreeCAD #22765: Constraint point on object not showing in the sketch toolbar

richiephp, 2025-07-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm not able to find icons, mainly the constraint point of object and another one that looks like an X with the dot in the middle

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.35
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-26:
Please use the forum for help requests.

closed 2025-07-26 by maxwxyz

richiephp, 2025-07-26:
No problem, found that they keep changing things, which is confusing