FreeCAD/FreeCAD #22768: Active Assembly causes cosntant stuttering

BackwardEcho, 2025-07-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When my assembly is active, there is a persistent and noticeable stutter in FC. Also when my assembly is active, I am unable to remove that asterisk ( * ) from the filename signifying there has been a change to the file even though I have just saved. It still seems to save properly, but FC thinks there has been a change made immediately after saving. Maybe they are linked?

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/32.512/1.25
Installed mods: 
  * AddonManager 2025.7.17
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * FusedFilamentDesign 0.25.250
  * Telemetry 1.0.2
```

### Subproject(s) affected?

Assembly

### Anything else?

@oursland asked to see if the recently fixed Windows dev release worked fine for people. At the time it did, but I am not sure if this isn't related to what he was asking about. Granted, the build listed in the `Full Info` is from `FreeCAD-Bundle`, but this issue happened in the regular Weekly release oursland asked about and is what prompted me to try the `FreeCAD-Bundle` version.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-07-28:
Was this stuttering happening in 1.0?

PaddleStroke, 2025-07-29:
I think https://github.com/FreeCAD/FreeCAD/pull/22823 is going to fix this issue.
If you can build the one line fix to test? Else wait for next weekly build to confirm

BackwardEcho, 2025-07-29:
> I think [#22823](https://github.com/FreeCAD/FreeCAD/pull/22823) is going to fix this issue. If you can build the one line fix to test? Else wait for next weekly build to confirm

If it was a file I could download and slap into my FC folder, sure. However I am having trouble finding where to download the dang thing and doing a search in said folder results in 0 hits for "AssemblyObjectPyImp.cpp". So I will probably just have to wait.

PaddleStroke, 2025-07-30:
It's a change in cpp so you can't copy paste it, it needsto be compile freecad from source.
However you can actually test if it is https://github.com/FreeCAD/FreeCAD/pull/22711 that is causing this bug because that PR was in python.

- Locate the file src/Mod/Assembly/CommandCreateSimulation.py
- Change isActive function back to the original. It that was : 
```
    def IsActive(self):
        return (
            UtilsAssembly.isAssemblyCommandActive()
            and UtilsAssembly.assembly_has_at_least_n_parts(1)
        )
```
- Save the file
- Restart freecad

BackwardEcho, 2025-07-30:
[image]

[image]

Thankfully I can edit it just fine with notepad. 

It worked. I no longer have stuttering and can properly save again (not that I lost the ability to save, just now FC won't think I need to save again).

closed 2025-07-30 by maxwxyz