FreeCAD/FreeCAD #22769: Sketcher: Constraint visibility toggle button doesn't work

ErtugrulEge, 2025-07-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!
In sketch mode the eye icon button which control visibility of constraints doesn't work.

In 1.0 when I click this button that hides all constraints as expected.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-27:
@AjinkyaDahale FYI

kaiwas, 2025-07-27:
I want to add that the checkboxes don't work for me. I can't turn them off as shown in the picture earlier.

https://github.com/user-attachments/assets/43c4ed9e-e81e-411f-9876-4cae0fee179a

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42721 (Git) AppImage
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2025.5.20
  * FreeCAD_SketchArch-master
  * PitchedRoof
  * Test
  * sheetmetal
  * addFC-main 1.5.1
  * sheetmetal.backup1752914681.7783923 0.7.24 (Disabled)
  * Road 2025.2.22
  * Reinforcement
```

closed 2025-08-09 by chennes