FreeCAD/FreeCAD #22774: FEM: nglib.h file not found when compiling with netgen truned on

muezabdalla, 2025-07-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

when trying to compile with netgen turned on I get this error:
```
/home/muez/applications/compiled/freecad/FreeCAD-git/src/3rdParty/salomesmesh/inc/NETGENPlugin_Mesher.hxx:42:10: fatal error: nglib.h: No such file or directory
   42 | #include <nglib.h>
```

### Full version info

```shell
OS: Void Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.42736 (Git)
Build date: 2025/07/26 20:46:49
Build type: Unknown
Branch: main
Hash: 91641c4fb600381b645ed55e568e6c3d4f14f39f
Python 3.13.5, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.1, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.38
  * AnimationFreeCAD 1.0.0
  * FreeCAD_Assembly4.1 0.50.18
  * ExplodedAssembly
  * CfdOF 1.31.4
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-27:
@FEA-eng any idea?

muezabdalla, 2025-07-28:
> What version of netgen is installed [@muezabdalla](https://github.com/muezabdalla) ?

I did not know that it needs netgen
because it did not give any errors when configuring so I thought it will compile it with smesh or something like that

closed 2025-09-22 by muezabdalla