FreeCAD/FreeCAD #22777: DXF: Error when imorting DXF files

semhustej, 2025-07-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- `File` -> `Import`
- Select any DXF file (I tested with 4 files)
- before the import options window opens up, error is shown. The error seems to have no influence on the functionality

```
17:34:49  Cannot find icon: preferences-cam
17:34:49  Traceback (most recent call last):
17:34:49    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-42721-conda-Windows-x86_64-py311\Mod\CAM\Path\Main\Gui\PreferencesJob.py", line 40, in __init__
17:34:49      self.form = FreeCADGui.PySideUic.loadUi(":preferences/PathJob.ui")
17:34:49                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:34:49    File "<string>", line 4, in <module>
17:34:49  RuntimeError: Cannot open file
17:34:49  JobPreferencesPage is not a preference page
17:34:49  Traceback (most recent call last):
17:34:49    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-42721-conda-Windows-x86_64-py311\Mod\CAM\Path\Base\Gui\PreferencesAdvanced.py", line 35, in __init__
17:34:49      self.form = FreeCADGui.PySideUic.loadUi(":preferences/Advanced.ui")
17:34:49                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:34:50    File "<string>", line 4, in <module>
17:34:50  RuntimeError: Cannot open file
17:34:50  AdvancedPreferencesPage is not a preference page
```

See behavior in the video:

https://github.com/user-attachments/assets/6afa3863-725e-4075-a0fb-6f939830e07d

The issue is not older than 4 weeks.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-27:
@furgo16 FYI

furgo16, 2025-07-27:
That's not related to the DXF importer, it appears every time you open the Preferences dialog. It is caused by CAM code. There is an open issue for it that I currently cannot find.

semhustej, 2025-07-27:
I can confirm the error is related to starting `Preferences` dialogue. (`Edit` -> `Preferences` from the menu).

furgo16, 2025-07-27:
Still happening for me:

```python-traceback
Cannot find icon: preferences-cam
Traceback (most recent call last):
  File "/snap/freecad/1598/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 40, in __init__
    self.form = FreeCADGui.PySideUic.loadUi(":preferences/PathJob.ui")
  File "<string>", line 4, in <module>
RuntimeError: Cannot open file
JobPreferencesPage is not a preference page
Traceback (most recent call last):
  File "/snap/freecad/1598/usr/Mod/CAM/Path/Base/Gui/PreferencesAdvanced.py", line 35, in __init__
    self.form = FreeCADGui.PySideUic.loadUi(":preferences/Advanced.ui")
  File "<string>", line 4, in <module>
RuntimeError: Cannot open file
AdvancedPreferencesPage is not a preference page
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42727 (Git) Snap 1598
Build date: 2025/07/25 22:54:43
Build type: Release
Branch: main
Hash: ce9bc869086159a8655e262500f00bd39cb4d860
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Launcher
```

maxwxyz, 2025-07-28:
@furgo16 so it is not a duplicate or fixed now?

furgo16, 2025-07-28:
Still reproducible on today's daily build. I don't even use the CAM workbench. @Syres916 so you're not experiencing the issue anymore?

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42742 (Git) Snap 1603
Build date: 2025/07/27 22:44:12
Build type: Release
Branch: main
Hash: f2e9a7f62eb1fc53db84db4a2d314e4f0a8d0734
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Launcher
```

Syres916, 2025-07-29:
> so you're not experiencing the issue anymore?

I definitely could open Preferences error free immediately on FreeCAD loading last week, I don't close Issues lightly but of course I didn't follow this exact reported procedure with regard to the DXF import. So I've scrapped my build folder and I'm compiling the latest git revision from scratch as I write this, will report back in a couple of hours.

furgo16, 2025-07-29:
Thank you. Note that the steps to reproduce are simply to open Edit > Preferences and are not specific to the DXF importer.

semhustej, 2025-07-29:
I can reproduce the issue with current weekly. I can reproduce it in safe mode too.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25

```

furgo16, 2025-08-04:
> so it is not a duplicate or fixed now?

@maxwxyz to summarize, the original issue was mistakenly closed without a fix. So this issue is technically a duplicate.

To keep things easy though, I'd leave this issue as it is. However, it might be worth considering adding the https://github.com/FreeCAD/FreeCAD/labels/Type%3A%20Regression and https://github.com/FreeCAD/FreeCAD/labels/Blocker labels.

Also, I'm not sure whether the affected project should be `Core` for what appears to be a bug in `CAM`, but I'm not 100% sure.

ErtugrulEge, 2025-08-09:
This issue is not relevant with DXF 

https://github.com/user-attachments/assets/831e1ba4-0947-474f-a597-26fa42ba995a

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61

```

furgo16, 2025-08-09:
@maxwxyz this issue is harmless, but disruptive to users. Should its priority be raised?

FEA-eng, 2025-08-11:
There's also this when clicking Ok in the Preferences:

[image]

semhustej, 2025-08-11:
> There's also this when clicking Ok in the Preferences:
> <img alt="Image" width="310" height="146" src="https://private-user-images.githubusercontent.com/59876896/476469008-b098361c-b09b-4052-89e7-4218377d0e55.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ4OTkwMTEsIm5iZiI6MTc1NDg5ODcxMSwicGF0aCI6Ii81OTg3Njg5Ni80NzY0NjkwMDgtYjA5ODM2MWMtYjA5Yi00MDUyLTg5ZTctNDIxODM3N2QwZTU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODExVDA3NTE1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4MjZkZmM5ZGQ2NTY2Zjk1MWFiY2YxZGEyZDdjNGJiMWM4MGEwYWQ4ODY2NGQxOGE3MDk3OTBjOTIwNDc4ZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2fKMZj0n6W9aAmmzoXHXndgx59424MXiTfID-1XvgmI">

I am not able to reproduce this behavior.

FEA-eng, 2025-08-11:
> I am not able to reproduce this behavior.

I have it all the time even in Safe Mode. Maybe it's Windows specific or you are using a different weekly build?

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

ErtugrulEge, 2025-08-11:
@FEA-eng can you try with this link ? I dont know why but now we have 2 type of weekly builds

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds 

https://github.com/user-attachments/assets/22e805b6-6283-4f92-8b7d-777048cd6a97

FEA-eng, 2025-08-11:
> [@FEA-eng](https://github.com/FEA-eng) can you try with this link ? I dont know why but now we have 2 type of weekly builds
> 
> https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

It's the same with that build but it seems to happen only when using the Safe Mode. Can you try that?

This weekly will be removed in the future and only https://github.com/FreeCAD/FreeCAD/releases will stay.

semhustej, 2025-08-11:
> There's also this when clicking Ok in the Preferences:
> <img alt="Image" width="310" height="146" [...]

I can confirm this behavior when testing in Safe Mode.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25

```

ErtugrulEge, 2025-08-11:
I can confirm with safe mode
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

https://github.com/user-attachments/assets/31cb2928-75dd-477a-86ef-205b031977e7

luzpaz, 2025-09-16:
The CAM preferences error shows up in #23949 as well