FreeCAD/FreeCAD #22779: DXF: C++ DXF import text with wrong height

semhustej, 2025-07-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When importing DXF using the C++ importer, the imported text heights (both dtext and mtext objects) are wrong.

I have a sample DXF file created in BricsCAD (attached to this post) with both Text and MText objects with sizes 100 and 200. The use a Standard text style with no fancy settings. The units in DXF are set up to be mm.

[FC-text-dxf-import.zip](https://github.com/user-attachments/files/21454930/FC-text-dxf-import.zip)

The resulting text sizes (both Text and MText) are 35.28mm and 70.56mm. There seems to be 0.3528 multiplication conversion done to the text height during the import process.
I have text scale multiplier set to 1.0 in FreeCAD

Text in BricsCAD:

[image]

Text in FreeCAD:

[image]

I can reproduce the issue in FreeCAD 1.0.1

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

None

### Anything else?

I got no answer in the forum: https://forum.freecad.org/viewtopic.php?t=97260

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-27:
@furgo16 FYI

furgo16, 2025-07-27:
I believe my last merged PR only fixed rotation (https://github.com/FreeCAD/FreeCAD/pull/22045/commits/e1420b43fb87be11d2fc82d6a5bb2298e298ccc5), but did not touch sizes.

It would be interesting to try whether https://github.com/FreeCAD/FreeCAD/pull/22251 fixes this.