FreeCAD/FreeCAD #22789: Gui: Expression input dialog for VarSets does simplify

pieterhijma, 2025-07-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The expression input dialog for VarSets does a simplify on expression before it is stored in the VarSet. Based on #22766, I doubt whether the dialog should do that. I now think, consistency is more important and it should store the raw expression without simplifying it, just as the regular expression dialog does things.

Below an example. Note that the example is made with https://github.com/FreeCAD/FreeCAD/pull/22788 because in `main` it currently crashes. Note that the VarSet does not store `sqrt(100)` but stores the simplified expression `10.00 mm`. In my opinion, if the user gives as input `sqrt(100)`, then that is probably what the user wants to have stored.

https://github.com/user-attachments/assets/3860f617-ecc4-4400-92dd-28096aec933e

### Full version info

```shell
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.42745 (Git)
Build date: 2025/07/28 08:17:18
Build type: Unknown
Branch: main
Hash: fcd1b923b387d9caa9dbadc28d3cc8af03a43b9c
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-07-28:
> then that is probably what the user wants to have stored.

Yes. You are absolutely right.

But it would also be great to have a convenient way to "destroy/realize" expressions. https://github.com/FreeCAD/FreeCAD/issues/14605

pierreporte, 2025-08-03:
The user input shouldn’t be changed so keeping the raw expression is a must. There could be a way to simplify it if needed and even destroying them as @kaiwas says though. But don’t change user input by default.