FreeCAD/FreeCAD #22807: TechDraw: dragging a view is not unodable

theo-vt, 2025-07-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Basically the title, you can undo a balloon annotation drag for instance, but not a view drag

https://github.com/user-attachments/assets/202560e1-ebab-4f8d-9b38-5a8d9ecec9da

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42714 (Git)
Build date: 2025/07/24 18:04:47
Build type: Unknown
Branch: document_transaction
Hash: 39ae8d1ddc5d49d0f3201cafd905426a0d2e0c52
Python 3.13.3, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * fasteners 0.5.39
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-29:
@WandererFan have you seen this before ?

WandererFan, 2025-07-29:
Never occurred to me to undo a drag. I just move the view back to where I want it.