FreeCAD/FreeCAD #22809: QT access results in crash

schmidtw, 2025-07-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This might be related to #22726, but the crash dump is different.  I imported a somewhat complex STEP file & spent about 30 minutes turning bits on/off working on figuring out what changes I wanted to make & out of the blue freecad crashed.

This happened twice & the traces are similar.  I rebuilt with debug symbols between the first and the second and got additional interesting data.

The top of the stack (excluding glibc) is in qlist.h:

1.
```cpp
    static constexpr qsizetype maxSize() { return Data::maxSize(); }
    qsizetype size() const noexcept { return d->size; } // <<--- this line, so d is null most likely.
    qsizetype count() const noexcept { return size(); }
    qsizetype length() const noexcept { return size(); }
```
2. https://github.com/FreeCAD/FreeCAD/blob/8dc6397fc4433785f73f8a1958ff11243c348376/src/Gui/Tree.cpp#L4788
3. https://github.com/FreeCAD/FreeCAD/blob/8dc6397fc4433785f73f8a1958ff11243c348376/src/Gui/Tree.cpp#L3535 is the next line to execute.  It looks like `pos->second` might be `null`

I'm adding a check and a debug statement & getting back to my project & will comment on this issue with any findings.

It seems like lists can be returned with null values in them.

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR42679 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
```

### Subproject(s) affected?

None

### Anything else?

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x19c30) [0x7fe017827c30]
#1  0x7fe01c438190 in Gui::DocumentItem::updateSelection(QTreeWidgetItem*, bool) from /home/schmidtw/projects/FreeCAD/build-g/lib/libFreeCADGui.so+0
#2  0x7fe01c43824b in Gui::DocumentItem::updateSelection(QTreeWidgetItem*, bool) from /home/schmidtw/projects/FreeCAD/build-g/lib/libFreeCADGui.so+0xbb
#3  0x7fe01c44975f in Gui::TreeWidget::onItemSelectionChanged() from /home/schmidtw/projects/FreeCAD/build-g/lib/libFreeCADGui.so+0x49f
#4  /lib64/libQt6Core.so.6(+0x1657ba) [0x7fe017f657ba]
#5  /lib64/libQt6Widgets.so.6(+0x40c89d) [0x7fe01960c89d]
#6  /lib64/libQt6Core.so.6(+0x1657ba) [0x7fe017f657ba]
#7  0x7fe0181a0633 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /lib64/libQt6Core.so.6+0x43
#8  0x7fe0181a8377 in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&) from /lib64/libQt6Core.so.6+0xc7
#9  0x7fe0181ab2d3 in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /lib64/libQt6Core.so.6+0x283
#10  0x7fe019603cc4 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /lib64/libQt6Widgets.so.6+0x7f4
#11  0x7fe0196046c3 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /lib64/libQt6Widgets.so.6+0x2e3
#12  0x7fe01956f5ab in QAbstractItemView::mouseMoveEvent(QMouseEvent*) from /lib64/libQt6Widgets.so.6+0x36b
#13  0x7fe0192a6908 in QWidget::event(QEvent*) from /lib64/libQt6Widgets.so.6+0x2b8
#14  0x7fe0193010f6 in QFrame::event(QEvent*) from /lib64/libQt6Widgets.so.6+0x36
#15  0x7fe017ef9caa in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x8a
#16  0x7fe01923d96f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x7f
#17  0x7fe019248593 in QApplication::notify(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0xef3
#18  0x7fe01c1983e0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/schmidtw/projects/FreeCAD/build-g/lib/libFreeCADGui.so+0xa0
#19  0x7fe017ef9f38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x168
#20  0x7fe019246e29 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib64/libQt6Widgets.so.6+0x5f9
#21  /lib64/libQt6Widgets.so.6(+0xbc3b5) [0x7fe0192bc3b5]
#22  /lib64/libQt6Widgets.so.6(+0xbeeb3) [0x7fe0192beeb3]
#23  0x7fe01923d97f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x8f
#24  0x7fe01c1983e0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/schmidtw/projects/FreeCAD/build-g/lib/libFreeCADGui.so+0xa0
#25  0x7fe017ef9f38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x168
#26  0x7fe0186f22a9 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib64/libQt6Gui.so.6+0x3d9
#27  0x7fe01875c76c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Gui.so.6+0xac
#28  /lib64/libQt6XcbQpa.so.6(+0x3a276) [0x7fe004af4276]
#29  /lib64/libglib-2.0.so.0(+0x40880) [0x7fe0140eb880]
#30  /lib64/libglib-2.0.so.0(+0x497c8) [0x7fe0140f47c8]
#31  /lib64/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fe0140f4973]
#32  0x7fe01820f9ad in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Core.so.6+0x5d
#33  0x7fe017f07b03 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Core.so.6+0x1a3
#34  0x7fe017f03419 in QCoreApplication::exec() from /lib64/libQt6Core.so.6+0xa9
#35  0x7fe01c0bb3a4 in Gui::Application::runApplication() from /home/schmidtw/projects/FreeCAD/build-g/lib/libFreeCADGui.so+0xaf4
#36  ./bin/FreeCAD() [0x4022b6]
#37  /lib64/libc.so.6(+0x35f5) [0x7fe0178115f5]
#38  /lib64/libc.so.6(__libc_start_main+0x88) [0x7fe0178116a8]
#39  ./bin/FreeCAD() [0x402645]

#1
eu-addr2line -f -C -e lib/libFreeCADGui.so 0x638190

QList<QTreeWidgetItem*>::size() const inlined at /usr/include/qt6/QtWidgets/qtreewidget.h:151:61 in Gui::DocumentItem::updateSelection(QTreeWidgetItem*, bool)
/usr/include/qt6/QtCore/qlist.h:428:49

#2
eu-addr2line -f -C -e lib/libFreeCADGui.so 0x63824b

Gui::DocumentItem::updateSelection(QTreeWidgetItem*, bool)
/home/schmidtw/projects/FreeCAD/src/Gui/Tree.cpp:4788:49

#3
eu-addr2line -f -C -e lib/libFreeCADGui.so 0x64975f

Gui::TreeWidget::onItemSelectionChanged()
/home/schmidtw/projects/FreeCAD/src/Gui/Tree.cpp:3535:28

#18
eu-addr2line -f -C -e lib/libFreeCADGui.so 0x3983e0

Gui::GUIApplication::notify(QObject*, QEvent*)
/home/schmidtw/projects/FreeCAD/src/Gui/GuiApplication.cpp:93:40

#24
eu-addr2line -f -C -e lib/libFreeCADGui.so 0x3983e0

Gui::GUIApplication::notify(QObject*, QEvent*)
/home/schmidtw/projects/FreeCAD/src/Gui/GuiApplication.cpp:93:40

#35
eu-addr2line -f -C -e lib/libFreeCADGui.so 0x2bb3a4

tryRunEventLoop inlined at /home/schmidtw/projects/FreeCAD/src/Gui/Application.cpp:2319:24 in Gui::Application::runApplication()
/home/schmidtw/projects/FreeCAD/src/Gui/Application.cpp:2294:13
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kadet1090, 2025-08-12:
Does this happen on current stable?

schmidtw, 2025-08-12:
I don't know.  I have no way to reproduce this reliably.  I have run into so many random crashes that I run with debug symbols on so I can at least debug the issues as they come up.

closed 2026-02-18 by schmidtw

FEA-eng, 2026-02-18:
> I'm closing the issue because I don't have the time to try to build a unit test to reproduce the potential crash, but that's how to confirm and solve this issue.

@maxwxyz Perhaps the issue should still be investigated (and maybe reopened by someone else) as it's likely to still be relevant. Who could have a look at this code? @kadet1090 maybe?

FEA-eng, 2026-02-18:
Thanks for the immediate reaction.

> We can create Code Quality issue to think about better way of managing document references so they don't become stale maybe

Sounds perfectly sufficient for now, I just want to make sure such code issues are not lost with reports like this one being closed.