FreeCAD/FreeCAD #22810: PartDesign Boolean messes up whole hierarchy/history of operations

razorree, 2025-07-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a 'bottom' part and I want to boolean cut it from top part. Top part has a binder to bottom, so I can use bottom part dimensions in my sketch to create top part.

[image]

[image]
I want to cut bottom part (or binder) from top (Pad001).
When I select 2 of them (Binder and Pad001) and press Boolean
I have:

[image]
When I try to remove one item from Boolean parameters (selecting in and pressing DEL key), everything gets even worse - in Model view, Boolean has more duplicated childred every time I try to remove something from "boolean parameters pane", also FreeCad starts to use a lot of CPU and freeze for 30-60 seconds.
After removing only 2 positions from "boolean parameters":

[image]

(**.txt** added to file extension)
[interface test1-before boolean.FCStd.txt](https://github.com/user-attachments/files/21477685/interface.test1-before.boolean.FCStd.txt)

### Full version info

```shell
OS: Ubuntu 25.04 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * AddonManager 2025.6.29
  * CurvedShapes 1.0.13
  * Curves 0.6.64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

razorree, 2025-07-28:
yes, I understand I used boolean not in a proper way, finally I got my head around it.
However, still, FreeCad behaviour could be a bit more predictable and more reasonable....

razorree, 2025-07-29:
the same behaviour in:
**FreeCAD_weekly-2025.07.29-Linux-x86_64-py311.AppImage**   

OS: Ubuntu 25.04 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * AddonManager 2025.6.29
  * CurvedShapes 1.0.13
  * Curves 0.6.64

and the same in **FreeCAD_weekly-builds-42721-conda-Linux-x86_64-py311.AppImage **
OS: Ubuntu 25.04 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42721 (Git) AppImage
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * AddonManager 2025.6.29
  * CurvedShapes 1.0.13
  * Curves 0.6.64

FEA-eng, 2025-10-18:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FlachyJoe, 2025-10-18:
The boolean should not allow selecting a function of the active body as a tool, except a binder.

FEA-eng, 2026-01-18:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260113 (Git shallow)
Build date: 2026/01/13 22:48:50
Build type: Release
Branch: (HEAD detached at 4cc0a1b)
Hash: 4cc0a1b718735be10261d936066cf1382dd51884
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

luzpaz, 2026-09-11:
>The boolean should not allow selecting a function of the active body as a tool, except a binder.

@FlachyJoe is this still accurate ?

FlachyJoe, 2026-09-12:
@luzpaz It seems to work now, at least in this simple case

[image]

But a better workflow would be
- Add an additive feature to subtractive feature transformation function
- Allow negative shape ie. a shape which acts as a subtraction when it fuse with something else. So we could start a body with a subtractive feature.