FreeCAD/FreeCAD #22814: Assembly: "a grounded object moved" if grounded object is not at (0, 0, 0)

jffmichi, 2025-07-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[assembly_fail.zip](https://github.com/user-attachments/files/21479353/assembly_fail.zip)

In the provided file I moved the grounded object to z = -10. This causes a lot of the following error messages when moving the sliding object:
```
04:32:20  Assembly : Ignoring bad solve, a grounded object (assembly_fail#Body_Slider_Link) moved.
04:32:20  Assembly : Ignoring bad solve, a grounded object (assembly_fail#Body_Slider_Link) moved.
04:32:20  Assembly : Ignoring bad solve, a grounded object (assembly_fail#Body_Slider_Link) moved.
```
Maybe related to #14444 and #14549. Afaict, the error message comes from that PR but the underlying problem seems to be that the grounded object is moved in the first place.

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42705 (Git)
Build date: 2025/07/25 02:00:45
Build type: Unknown
Branch: main
Hash: 705f2daa61b2cab2a3cb4885ac206bb775232af5
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc1-22d437b771
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/breeze
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7555/1.5
Installed mods: 
  * kicadStepUpMod 11.7.1
  * Gridfinity 0.11.9
  * sheetmetal 0.7.22
  * fcgear 1.0.0
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Curves 0.6.62
  * lattice2 1.0.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-30:
@PaddleStroke FYI

PaddleStroke, 2025-07-31:
issue description is not correct, it's not that grounded objects must be at origin. It's that you cannot move grounded objects with transform tool or by modifying their placement directly.

I need to think on the best way to solve this.

PaddleStroke, 2025-07-31:
Actually with the sample file even creating a grounded joint is problematic. There's something fishy going on I can't pinpoint it yet but sometimes it works sometimes it prints the error. I wonder if this is file specific. Did you made this file in 1.1?

PaddleStroke, 2025-10-20:
@maxwxyz