FreeCAD/FreeCAD #22815: Assembly: active assembly messes with ui focus (rename and hover not working)

jffmichi, 2025-07-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

(not sure how to properly name this issue)

When an assembly is active, renaming an object using F2 isn't working any more, the rename operation is aborted instantly. Also hovering an object highlights it only for a split second in the 3d view.

The following video shows the normal behaviour without active assembly, then activates the assembly and shows the issue.

https://github.com/user-attachments/assets/162a26bc-6eea-40b4-b877-eccbec6ae836

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42781 (Git)
Build date: 2025/07/27 16:48:29
Build type: Release
Branch: makepkg
Hash: 8feddbca339b6555a0b6edce7febd8661156b07d
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc1-22d437b771
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/breeze
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7555/1.5
Installed mods: 
  * kicadStepUpMod 11.7.1
  * Gridfinity 0.11.9
  * sheetmetal 0.7.22
  * fcgear 1.0.0
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Curves 0.6.62
  * lattice2 1.0.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-29:
cc @PaddleStroke

PaddleStroke, 2025-07-29:
Very likely related to https://github.com/FreeCAD/FreeCAD/issues/22768

PaddleStroke, 2025-07-29:
After bisecting I finally found that it comes from the most unlikely commit : https://github.com/FreeCAD/FreeCAD/pull/22711

closed 2025-07-29 by chennes