FreeCAD/FreeCAD #22818: Core: Remove Yes/No text from the boolean properties

semhustej, 2025-07-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request:**

**Current situation:**
Currently (FreeCAD 1.1dev), boolean properties have two states in the properties panel. Checked checkbox and Yes text or unchecked checkbox and No text.

This can be confusing, as the text seems to be related to the checkbox. This does not logically make sense as there are two states:

1.

[image]
Logically this means that Yes is checked. This makes sense.

2.

[image]
Logically this means that No is unchecked. That means **Yes**, because No is disabled (not checked). This can be confusing.

**Proposed solution**
Remove the text (Yes/No) completely. Checkbox is enough for users to understand whether the boolean is set to true or false.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Core

### Anything else?

I posted a short video showcasing the new feature and 3 users reacted in the same way - remove the text:
https://www.reddit.com/r/FreeCAD/comments/1mbil8e/ui_of_freecad_is_getting_better_every_day/

[image]

[image]

Other software tools don't mix text and checkbox with boolean switcher:
Revit - checkbox

[image]

AutoCAD - text:

[image]

Blender - checbkox:

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-29:
@FreeCAD/design-working-group what about this UI request?

kadet1090, 2025-07-29:
Agree, but I'm not sure if we should remove the label altogether. Maybe just change it to static "enabled"?

obelisk79, 2025-07-29:
I agree with changing the text to enabled.

kaiwas, 2025-07-30:
> changing the text to enabled

and what text will be in the off state? Disabled?

is the text really needed at all?
and why can't it be true/false?

semhustej, 2025-07-30:
> Agree, but I'm not sure if we should remove the label altogether. Maybe just change it to static "enabled"?

I would vote against this option. Checkbox is enough in my opinion.

- checkbox is universally understood to enable/disable something.
- when I look at the screenshot from revit in the OP (only checkboxes), I see no confusion at all
- adding `Enabled` text adds to information overwhelm (more text) in the properties panel. I think less text that user has to perceive in the properties panel will lead to better UX.
Look at this sample comparison with few boolean properties. To me it is more concise without the text:

[image]

Properties with text

[image]

> but creating checkboxes without label is also not great due to small click area

- the target area here is the whole cell, not the check box only. So having text there makes no difference. At least that is what my test shows.

[image]

- I don't think there is a need to reinvent the wheel. Can someone post an example of another software package, where there is a checkbox + text in a similar situation? I tried to search but could only find checkbox only (Revit, Scribus, Fusion 360) or text only  (AutoCAD, BricsCAD, BIMvision) plus Blender, but that is a bit of a different setup. I found no software package where checkbox + text is combined in similar setup.

> I would also like to show the state as text (in terms of accessibility features)

- this is a valid reason to leave the text there. I don't know much about accessibility features so I am not sure how helpful this is and if it should override the above mentioned points.

kadet1090, 2025-07-30:
> Why then are there so many boolean checkboxes in the program without additional captions? )))

Because FreeCAD is anything but good at having consistent and proper UI/UX. The "box" part of the widget should be aligned to top left of the label and first line should be roughly centered with the box (baseline of the tickmark and text should be the same). Both should be clickable and both are important. Having the box in any other place is to put it simply, an mistake in terms of UI design - and FreeCAD has a ton of different mistakes. 

> the target area here is the whole cell, not the check box only. So having text there makes no difference. At least that is what my test shows.

That is not enough, as users will not know it - you don't expect the whole cell to be clickable and to toggle the state - only label and checkbox.

> > I would also like to show the state as text (in terms of accessibility features)
>
> this is a valid reason to leave the text there. I don't know much about accessibility features so I am not sure how helpful this is and if it should override the above mentioned points.

You should not have label changing with checkbox, because the checkbox itself reads as "yes" or "no". So you in no circumstances should have dynamic label for checkbox as it would be confusing - as the reddit thread shows. I even noted in in the original PR: https://github.com/FreeCAD/FreeCAD/pull/21555#issuecomment-2935500358. With simple "yes" and "no" I hoped that it would however be clear enough and give a bit of a hint of the current state that is easier to process. The reddit thread shows that it is not.

The problem here however is that the checkbox has label but it is placed on the left of the "box" instead of the usual right. I'm fine with removing the label all together - especially that this kind of widget is kind of standard in the industry. From UI/UX point of view it is still a mistake to create checkboxes in that way, but at least it cannot be confusing that way.

kaiwas, 2025-07-30:
> Because FreeCAD is anything but good at having consistent and proper UI/UX.

I agree.
but if there are already too many holes - one more hole won't make things much worse.
A fly in the ointment or two fly in the ointment - the difference is not great.
Why is it that in this particular issue we should fight so hard for design, while in other places there is no improvement? )

kadet1090, 2025-07-30:
> I still haven't been told the "true/false" flaw, as it was in the drop-down list. How is this option worse than yes/no?

It is exactly the same if you put it with checkbox.

belinski-roman, 2026-10-15:
> 20\. I agree, all Yes and No texts should be removed. Just empty or filled checkboxes as default enough, just like Blender and other programs that showed above posts
>      <img alt="Image" width="371" height="238" src="https://private-user-images.githubusercontent.com/146654654/539292398-85426bbb-c268-406b-884d-7319fd2d372e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTIwOTA2NDgsIm5iZiI6MTc5MjA5MDM0OCwicGF0aCI6Ii8xNDY2NTQ2NTQvNTM5MjkyMzk4LTg1NDI2YmJiLWMyNjgtNDA2Yi04ODRkLTczMTlmZDJkMzcyZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYxMDE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MTAxNVQxODUyMjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NjExZTQwNGQwYmE0NDk0NzhkMGY0MmZmMWE3MGZkNGU0NmEyYjY4NjBlMDRkZWZmZWIzNzhhNmQwZDY0NGNhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZwbmcifQ.THZ7RhBtt1iIjj4Odavbi3nneNulbIcSS6v1JEC-c1s">

The gold standard