FreeCAD/FreeCAD #22822: BIM: Slab Face Maker settings does not affect the object

semhustej, 2025-07-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When `BIM_Slab` is created from sketch, `Face Maker` settings does not work. It seems as if `Cheese` is set no matter what the `Face Maker` settings in Slab properties is.

See sample file with slabs created from the same sketch shape with these Face Maker settings: None, Simple, Cheese, Bullseye:

[image]

This is how the sample file looks like in FC 1.0.1:

[image]

Sample file (remove the zip extension): 
[FC_slab_facemaker.FCStd.zip](https://github.com/user-attachments/files/21490533/FC_slab_facemaker.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

The same issue can be observed in `Arch_Structure` tool, which is the parent command of the `BIM_Slab` tool.

I am not sure if this is not a feature of FC 1.0.1, but if it is, the `Face Maker` property should be disabled/hidden.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-07-30:
> For me this is a questionable property. The face maker choice should be handled automatically and not be exposed to the user.

For `BIM_Slab` I agree. I can not think of a reason why user would want a different setting than `Bullseye`. This would make the tool more user friendly for sure.

I am not sure about `Arch_Structure` (as I don't know all the places it is used) but it seems to me the same could be applied.

markkomartin, 2025-07-30:
> > For me this is a questionable property. The face maker choice should be handled automatically and not be exposed to the user.
> 
> For `BIM_Slab` I agree. I can not think of a reason why user would want a different `Face Maker` setting than `Bullseye`. This would make the tool more user friendly for sure.
> 
> I am not sure about `Arch_Structure` (as I don't know all the places it is used) but it seems to me the same could be applied.

I agree too. @paullee0  do you think there could be problems with SketchArch?

paullee0, 2025-07-30:
> Confirmed. Regression introduced with [#18336](https://github.com/FreeCAD/FreeCAD/pull/18336).

Thanks, indeed L980 seems need to be deleted.

https://github.com/paullee0/FreeCAD/blob/5735e337b1db28e2fdd1f4b2b185605080aaba27/src/Mod/BIM/ArchStructure.py#L980

`                            faceMaker = 'Bullseye'
`

> ... [@paullee0](https://github.com/paullee0) do you think there could be problems with SketchArch?

SketchArch provide its own information about FaceMaker, no problem is anticipated.

Thanks.

semhustej, 2025-07-31:
> For me this is a questionable property. The face maker choice should be handled automatically and not be exposed to the user.

Should I create a separate issue for this (changing the default Face Maker setting and hiding the property from user)?

paullee0, 2025-07-31:
Face Maker `Bullseye` works as expected
> 
>     * Face Maker 'Cheese' does not work (the result is the same as `None`) and throws this error:
> 
> 
> ```
> 19:56:46  class Standard_Failure Failed to validate broken face
> 19:56:46  Facemaker returned an error
> 19:56:46  class Standard_Failure Failed to validate broken face
> 19:56:46  Facemaker returned an error
> ```

No problems with AppImage on Linux Fedora40, the latest one also - FreeCAD_weekly-builds-42721-conda-Linux-x86_64-py31.Appimage

Just Cheese does not create opening for that rectangle touch one edge of the slab, when it is moved away from the edge, it creates an opening.  This is probably a Part issue on Windows?

[image]

furgo16, 2025-08-01:
Related forum post: https://forum.freecad.org/viewtopic.php?p=823756#p823756

semhustej, 2025-08-01:
> Just Cheese does not create opening for that rectangle touch one edge of the slab, when it is moved away from the edge, it creates an opening. This is probably a Part issue on Windows?

I did a test with a different dataset and the error is related to different issue. So my conclusion is that **the edited library works as expected**.
I am sorry for the confusion.

PR #22859 (2025-08-27): [ArchStructure]  Fix Regression : FaceMaker
Fix #22822

See discussion and example model file in github :- https://github.com/FreeCAD/FreeCAD/issues/22822

## Issues

## Before and After Images