FreeCAD/FreeCAD #22847: BIM: Column Position Z property doesn't work if multiple columns selected

ErtugrulEge, 2025-07-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!
1.Create some columns 
2.Select columns and change z property 
3.Other columns x,y,z positions will be same with first selected column

see behaviour:
https://github.com/user-attachments/assets/a928b27f-2131-4efc-87ce-5285f464a612

If there is a similar issue, can @kaiwas   help? because he has some similar issues 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-07-30:
When changing the properties of all objects, not only the Z changes, but the X/Y coordinates are also combined.

I would use the Move tool. Although I agree that this behavior of the properties is very strange.

semhustej, 2025-07-30:
I can confirm the behavior. It is not related to Column, it happens with other draft and BIM objects. It is a surprising behavior as only the changed Z coordinate of the selected objects should change in my opinion.

I can reproduce the issue in FC 1.0.1

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

furgo16, 2025-07-31:
This is one of the first bugs I stumbled on when I first started using BIM. A few tutorials showcase this as a FreeCAD feature: select multiple objects, then modify a property, and the change propagates to all objects. Yet when you try to do this with a component of the `Placement` property, it breaks in the way described in this issue. However, it is not specific to BIM, as far as I can tell. You can reproduce it following the same steps with two Part Design bodies, for instance.

Since then, I've learnt not to modify the placement in a selection of multiple objects, but I still think this is unexpected and non-consistent behaviour.

My guess is that the `Placement` property (`App:PropertyPlacement`) can only be written at once, instead of each of its components separately. Thus when e.g. the Z placement of the last selected object is modified and saved, the full placement (x,y,z, and probably rotation) overwrites the placement of all other selected objects.

semhustej, 2025-07-31:
In my opinion the change of the Z value should change the Z value (and only the Z value) of all selected objects.
But I guess this goes deep inside of the core FreeCAD code, as furgo16 points out.

markkomartin, 2025-07-31:
> > In my opinion the change of the Z value should change the Z value (and only the Z value) of all selected objects. But I guess this goes deep inside of the core FreeCAD code, as furgo16 points out.
> 
> This is the ideal solution but I don't know how it is difficult the programming such thing. At least, we can do it like Blender, easier and quick solution.

I confirm the bug with the columns. With the walls, everything works fine. 
For consistency, I should be able to change only the z-coordinates of various selected columns ( as @semhustej suggests ) .

I find it easier to use the move or transform tool ( transformation tool isn't available for multiple selected objects), but it doesn't change anything. I agree with the high priority of this bug.

semhustej, 2025-07-31:
> With the walls, everything works fine.

This is because the location of the wall is primarily set by the base object. By default all wall objects have X, Y and Z coordinates set to 0.

If you first change the X coordinate of one of the walls to e.g. 10m and than select both walls and change the Z coordinate to e.g. 5m, you will see that the first wall's X coordinate is changed without you adjusting it.

markkomartin, 2025-08-01:
> > In my opinion, a first step would be to make the coordinates box uneditable when I select multiple objects and the coordinates are different.
> 
> I disagree. I think you should be able to change the property in bulk - regardless of whether several selected objects have different values.

The final goal is what you suggest, but in my opinion it's not very easy to achieve. 
Disabling the coordinates box is better than the current situation. 
At least one user is encouraged to find an alternative solution (draft move).