FreeCAD/FreeCAD #22858: Preferences throws error build 42780

sslupsky, 2025-07-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When attempting to save preferences an error is thrown:
"The selected asset path is not writable."

If I change a preference (Camera Type), the change appears to stick when the preference pane is opened again and through restarting FreeCAD. 

To recreate:

Start FreeCAD
Select Preferences
Click "Ok"

[image]

### Full version info

```shell
OS: macOS 15.6
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/109/1
Installed mods: 
  * sheetmetal 0.7.11
  * Curves 0.6.58
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

sslupsky, 2025-07-31:
@luzpaz I went back and grabbed a copy of the old preferences from a backup and attached here if you want to try restoring these to recreate this issue and the Polyline issue reported in #22855 .

I scanned through them and it appears to me there is a corruption at line 816 in user.cfg.  I have no idea how the corruption occurred but I restored from a backup that shows the file was last modified on July 21.

I'll leave the issue open in case you want to track it further.  It might be worthwhile to consider what is done if a preferences file is found to be corrupted.  Otherwise feel free to close the issue.

[freecad preferences.zip](https://github.com/user-attachments/files/21537999/freecad.preferences.zip)

sslupsky, 2025-08-04:
@halfdane Creating the directory (~/Library/Preferences/FreeCAD/CamAssets) resolved the issue for me on macOS.  Thank you.

pmjdebruijn, 2025-08-28:
> Seems like the only "asset path" in the preferences are the CamAssets, and indeed creating the directory by copying it from the input field and running `mkdir -p ~/.config/FreeCAD/CamAssets` (Ubuntu 22.04) fixed the problem for me.

I have this issue on self built current git master as well, and the `mkdir` resolved the issue for me as well...

closed 2025-09-03 by chennes

ReneSci, 2025-09-24:
Yes, it is ok now

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * A2plus 0.4.68
  * CfdOF 1.32.1
  * Defeaturing 1.2.9
  * freecad.optics_design_workbench 0.6.2
  * Manipulator 1.6.2
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.2.2
  * QuickMeasure 2022.10.28
  * sheetmetal 0.7.52
  * toSketch 1.0.1
  * workfeature

```

commit 0020057f (2025-09-03): [CAM] Make sure CAMAsset location exists (#23618)

* Fixes #22858

Ensure that the default CAM asset location exists and is created in the user data location

* Update src/Mod/CAM/Path/Preferences.py

---------