FreeCAD/FreeCAD #22867: Assembly: Deletion of an object changes the rotation of other unrelated objects (even if they are grounded)

derMart, 2025-08-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

See the video and the attached file where the problem occurs. Deleting ProfilQuer3 results in the rotation of other objects (SolarPanelX) to be changed. Even if those objects are constraint by coincident joints and grounded joints. 
The following errors appear when deleting the object:
```
09:44:43  <PropertyLinks> PropertyLinks.cpp(447): solar_roof#Joint002.Reference1 missing element reference solar_roof#SolarPanel001 SolarPanel001.;#2b:1;:G;XTR;:H286:7,F;:H287,F.Face1
```

Also when one undo the delete the coincident joints are displayed still in the wrong position.

https://github.com/user-attachments/assets/705f9ef1-6e09-468d-98cf-079262e9d81e

[solar_roof.zip](https://github.com/user-attachments/files/21543810/solar_roof.zip)

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, OCC 7.9.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-01:
Please test 1.1dev to see if you can reproduce  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

derMart, 2025-08-01:
@luzpaz The rotation does not happen in 1.1-dev on recompute :-) But I do see the Fixed002 as broken link:

[image]

... and I just wonder if this is related to the topological renaming problem. It could have been the case that I made the joint before I added some holes to the bodies the joint connects ... and I also remember that there were open issues here related to the fact that the assembly workbench still has problems related to topological renaming?

luzpaz, 2025-08-01:
@PaddleStroke any chance you can take a look here ?

@derMart can you please add your About info to the 1.1dev comment above? TIA

drwho495, 2025-08-03:
> [@luzpaz](https://github.com/luzpaz) The rotation does not happen in 1.1-dev on recompute :-) But I do see the Fixed002 as broken link:
> 
> <img alt="Image" width="447" height="107" src="https://private-user-images.githubusercontent.com/1069456/473591556-33dd13b4-4b53-4cde-ba69-d5e62327f3e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQxODcyNDgsIm5iZiI6MTc1NDE4Njk0OCwicGF0aCI6Ii8xMDY5NDU2LzQ3MzU5MTU1Ni0zM2RkMTNiNC00YjUzLTRjZGUtYmE2OS1kNWU2MjMyN2YzZTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MDNUMDIwOTA4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmJjNjcyYTVjNzlkYjI4MzcyMWMwOTM0YjQ0YWNkZDQzZTMxZGU2MzJiMDFhZjQ0MTg4NDZmNDEwNDZmY2FhZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.h7zmSkm9n2Xy3YAKqkLpj2GfB3YDA78zXaaHW-XvG04">
> ... and I just wonder if this is related to the topological renaming problem. It could have been the case that I made the joint before I added some holes to the bodies the joint connects ... and I also remember that there were open issues here related to the fact that the assembly workbench still has problems related to topological renaming?
> 
> EDIT: No wait, it is different. Before I deleted ProfilQuer003 the unrelated joint Fixed002, which is joining two different bodies is not shown as broken in 1.1-dev, only after I delete ProfilQuer003 it is shown as broken link ... EDIT2: Even more strange. I can recompute the whole assembly and Fixed002 is not marked as broken, only if I delete ProfilQuer003 it will be marked broken or if I go into edit mode of the Reference1 of Fixed002 ...
> 
> About info on the dev version used here:
> 
> ```
> OS: Arch Linux (GNOME/gnome/wayland)
> Architecture: x86_64
> Version: 1.1.0dev.42721 (Git) AppImage
> Build date: 2025/07/25 17:44:16
> Build type: Release
> Branch: main
> Hash: b5d782e85690f12be44513e1f1e510b367a69c27
> Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
> shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: German/Germany (de_DE)
> Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
> Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
> Logical DPI/Physical DPI/Pixel Ratio: 96/169.808/1
> ```

this is because propertylinks are probably updating after deletion of an object. also this issue is known, and fixed, when a reference broke the assembly would blow up. if you can find more issues then please share. the toponaming issues may be fixed in a few weeks.

drwho495, 2025-08-03:
im able to run on my main branch (with like 15 toponaming fixes atp) and there are no issues, so some of the issue could be propertylinks related? and that bug might be fixed by complexFind/migration cache/the propertylinks fix

let me test on stock 1.1dev

drwho495, 2025-08-03:
also if you would like to improve tnp in assemblies right now, then enable allow compound on all of your parts

closed 2026-09-14 by maxwxyz