FreeCAD/FreeCAD #22871: UI/UX/Varset/Property View: Improve the readability of the property selection list when creating them.

kaiwas, 2025-08-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This problem may have already been discussed somewhere (I couldn't find it).

When assigning properties (in Warset or property view), a problem arises that can be briefly described with the words: "My eyes are bleeding". The list of available properties is very difficult to read due to constantly repeating parts in the property names.

My proposal is quite simple - make a grouping of properties and put the repeating properties in the group header.

The concept is in the picture.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42721 (Git) AppImage
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Curves.backup1751366017.4487264 (Disabled)
  * addFC-main 1.6.2
  * Reinforcement
  * Curves.backup1751365993.7542331 0.6.63 (Disabled)
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * FrameForge 0.1.3
  * Curves.backup1751366131.5540717 (Disabled)
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.64
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-01:
@FreeCAD/design-working-group FYI

pierreporte, 2025-08-02:
It could be a good idea.

Do those types have anything to help extracting relevant information?

obelisk79, 2025-08-03:
I'm not sure how I feel about this. I understand why you might like to see this, but I'm just not sure I ever viewed this as a problem. Let me bring this for discussion with the DWG on discord for more discussion. I'll post feedback here once we've come to a general consensus.

kaiwas, 2025-08-03:
> We don't recommend such a change at this time.

I wasn't surprised at all :)

kadet1090, 2025-08-03:
> For starters, it would be possible to "mask" this problem

Masking a problem is not a good solution, it can create more trouble and misunderstandings than it would actually help. FreeCAD already contains too much half-baked solutions and that is a problem. I do agree that having some improvement is better than having none, but it must be net positive change. Here the improvement is not big and it can potentially be misleading because we are hiding some information - so I'd say that it is net zero change. It basically changes one solution to another one with similar problems while not addressing the deeper issue.

MisterMakerNL, 2025-08-03:
@kaiwas I was just looking at this, and it seems like the list is a tree. But it is presented as a list.
Now I am much an idiot in QT, so wonder if this isn't a better way in presenting these parameters?
Taggin @pieterhijma for fyi