FreeCAD/FreeCAD #22878: Measureing angles don't work most of the time.

leoheck, 2025-08-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Measuring angles works in specific situations only.

We need something like the "distance free" for angles.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-08-02:
The errors reported in the image seem to point to different issues. Please attach an error-free file and give reproducible steps.

luzpaz, 2025-08-02:
cc @hlorus any chance you can have a look ?

leoheck, 2025-08-04:
Maybe this is related.

[image]

- It is missing the complementary side (I would like to have the measure of the other side, which is 140)
- The measure arc of the angle does not fit the right place.

3x380V, 2025-08-14:
@hlorus, could you verify mentioned commit and eventually open PR for it preferably adding
```
Fixes: f4c90c07f1c0061fe8ca067f51c39c4e5e56592f ("[Part] Add handlers for unified measurement facility")
```
to the commit description?

luzpaz, 2025-08-19:
@leoheck @hlorus any chance you can test #23208 ?

leoheck, 2025-08-19:
Sure, I am building it...

leoheck, 2025-08-26:
I am not sure if the latest build already fixed this but I still cannot measure angles like this.

[image]

```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.43071 (Git)
Build date: 2025/08/25 18:08:30
Build type: Release
Branch: main
Hash: d85613b498881c47485a7d930807673203086756
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```